df8oe/UHSDR #1: horizontal bar which shows recept bandwidth does not change correct

df8oe, 2015-09-09:
Hi Clint,
if I change from 20m (where USB is selected) to 40m (where LSB is selected), the bar remains on the right side showing USB receiption (but LSB is heard correctly). I I change once to USB and then back to LSB, the bar changes correctly. It seems so that setting of the bar is only made when changing modes but must be done when changing bands, too.

vy 73
Andreas

imoldovavan, 2015-09-13:
I think this is something I can fix. Would this be considered a "hotfix" branch? What is the proper way of starting this?

df8oe, 2015-09-13:
Yes - start a "hotfix-branch". In this state everything which demonstrates our team-power is highly welcome.

vy 73
Andreas

imoldovavan, 2015-09-13:
Understood. hotfix_bandwith_bar branch started. Once we get something working we will ask Clint to take a look at it. 

73's
Ion

df8oe, 2015-09-19:
I have fixed this today in my devel branch.

commit cb5d6d2d (2017-11-16): Freq ruler fix #1, wfall marker colour fix.
commit 413980a1 (2017-12-01): Merge pull request #1244 from DD4WH/active-devel

noise blanker display fix #1
closed 2015-09-19 by df8oe