df8oe/UHSDR #3: Stuck at splash screen.

imoldovavan, 2015-09-12:
I build the master branch with coocox and the radio keeps flashing the splash screen. It looks like it resets itself during init.

df8oe, 2015-09-12:
My radio and the radio of another OM runs without problems... Could you please test it with the binary included in master branch and tell me if this works for you? This binary was build on console and I compared it to binary resulted from Eclipse-building. I have not tested with coocox but I do not have changed anything related to coocox. If you tell me if binary works for you I can focus to the reason...

vy 73
Andreas

imoldovavan, 2015-09-12:
Ok. Uploading the firmware from the repo works. I have to figure out what is going on with coocox. Has anybody build it with coocox succesfully?? I have the 1.7.8 version of coocox which worked before for all my stuff. I will try the latest coocox see if that makes a difference.

Ion

df8oe, 2015-09-12:
Many thanks for your report. I will check out what is the problem.

I must change some declarations because they get errors in console and Eclipse - but this is all I changed. But it is important  that compiling using coocox does work, too. Otherwise Clint is "kicked off" - absolutely inaccaptable :( I will have a look at this. Compiling using 1.7.8 MUST work - newer versions are never successfully. Be back again with results (when I got some)...

vy 73
Andreas

imoldovavan, 2015-09-12:
I am able to build successfully older versions of the repo but not this one. I even tried different versions of the toolchain, 4.7 and 4.9 with no success. I tried this on two different computers, one of them with a new install of coocox 1.7.8 and does not work. Let me know if I can help in any other way.

73's
Ion

df8oe, 2015-09-13:
I have pulled master branch and build with coocox without problems. Firmware can be flashed and works. But I have noticed that coocox compiler optimization switch stands at "O3" when you open CoIDE configuration file. I never have had touched this because of I am working with Linux - I do not have idea how this switch changes state. Please have a look at this adjustment in CoIDE - it must be "None". To be found in "Configuration/Compile Optimization".

Feedback please!

vy 73
Andreas

imoldovavan, 2015-09-13:
OK, that did the trick. It works now. Learn something every day. I will close this issue.

closed 2015-09-13 by imoldovavan

df8oe, 2015-09-13:
OK. I have corrected master branch firmware.coproj file. I think at the last use of CoIDE I have changed this value by accident using my touchpad. I have not recognized this - sorry for the inconvenience.

vy 73
Andreas

imoldovavan, 2015-09-13:
No problem. I tried to do the same thing.  Sorry about that, hopefully I haven't screwed anything up.

df8oe, 2015-09-13:
All ok. That is why "issues" exist. We all do make mistakes and errors - but it's not necessary that others stuck into the same trap... Now repo is clean and functionally due to your issue-start.

vy 73
Andreas

imoldovavan, 2015-09-13:
How do we propagate this change to the other branches??

df8oe, 2015-09-13:
It's already committted in all branches.

vy 73
Andreas