df8oe/UHSDR #7: REQUEST: development settings save

Rescue9, 2015-10-11:
It would be nice if there were another menu function to save development specific settings that aren't usually saved in eeprom write. For example; ability to save Disp. Pwr. setting when _Dev. Settings_ variable is _on_. In addition, I think that if _Dev. Settings_ variable is turned off, it should remove all development settings variables from the eeprom so that they don't interfere with normal operation. Thoughts?

df8oe, 2015-10-11:
Dev Settings variable has another function in Clints fw. It enables debugging via serial line. Until now there are no thoughts to implement a devel variable. Devel branch might have functions that don't work but no functions that are "hidden" and only can be enabled in setting a variable. I think that is not neccessary for every coder works his own way and there is no common thinking about such thing UNTIL NOW. Maybe things change some day - but I don't think this in nearer future...

vy 73
Andreas

df8oe, 2015-12-06:
I do not know which settings you mean. Please give an example.

vy 73
Andreas

Rescue9, 2015-12-06:
For one, I would like to have the opportunity to have Disp. Pwr. set on all the time. IN addition, The CAT ability. Having a "developer save" option would allow these settings to be saved, but ONLY if the developer save option is on. This forces users to enter a "developer' mode to save such settings.While this doesn't seem too useful now, it might in the future as more features are added.

df8oe, 2016-01-30:
I don't think its neccessary at the moment....

closed 2016-01-30 by df8oe