df8oe/UHSDR #8: BUG: Spectrum scope overlays EEPROM Reset info.

Rescue9, 2015-10-11:
I pulled the latest "devel" today to troubleshoot my TX power issues. When resetting the eeprom by F1-f3-f5, the device did reset properly, but while reading the informational message, the spectrum scope overlaid the lower half after a few seconds. In my opinion, there should be a "pause" in startup until the user presses a button to give slow reading users time to fully read the message.

df8oe, 2015-10-11:
OK - I have tested and its the same here. I can dig into the code and fix it (like the "bandwidth-bar-bug") - but I don't want to do that. Please be patient - I will walk through the github-processes with Clint so that he can easily grab such reports and can learn how to branch for such a small bugfix. Maybe this can last a few days - but it is more important to get close to possibilities of Github than fixing this (small) problem in shortest time. Discussion between Clint and me will be "directly" for other readers of NG are not interested in such a discussion.

vy 73
Andreas

Rescue9, 2015-10-11:
No worries. I agree with properly fixing versus time frame. Small bugs should not be rushed. Thanks for looking into it.

cctbcn, 2015-10-12:
The version that I posted was mostly used to "test" the github. Being 
that I only had a bit more than an hour or so to knock out the addition 
of the "Load Defaults" function, I didn't get a chance to inhibit the 
initialization/display of the waterfall/spectrum scope - a fairly 
trivial thing to do - but I simply ran out of time, and I mostly wanted 
to see if the Git "Pull" would work on my development system.

There are, perhaps, a half-dozen steps required to fully disable the 
spectrum scope (and I'd disable the audio as well as tuning, transmit 
and menu system - and inhibits the "Load Default" feature if a new FW 
version is being used for the first time for obvious reasons!!!) - but 
the radio would be effectively disabled and the  - you MUST do ONE of 
the following:

- Disconnect the power from the radio, preserving your "old" settings.

OR

- Use the POWER button to turn off the radio to cause an EEPROM WRITE, 
overwriting your "old" settings in EEPROM with the now-loaded defaults.

It is intended that the radio will be "stuck" in that state until the 
user does one of the above and the on-screen message will be modified to 
make it clear that this is what the user must do:  I want to make sure 
that the user *really* wants to blow away all of the settings and not do 
so by accident!

Since I now know that the Git pulls work for me (although I'm not quite 
sure how to integrate other peoples' changes into everything just yet) I 
can produce a version that has the aforementioned issues resolved.

73,

Clint
KA7OEI

df8oe, 2015-10-13:
Hello Clint,

I think you go a very complicated way in implementing "loading defaults"...

I suppose this way:
- create a shortcut that mcHF does not boot to radio screen but into menu
(e.g. "F1-F3-F5)
- create a new menu point in "extended menu" called "factory reset"
- if this is choosen and confirmed radio starts with factory settings, and
storing during power-down is disabled now. If user wants to overwrite his
old settings he MUST do that with pressing menu for a longer time.

I have read the instructions very fast so that I do not have the issue :)
But it is correct that this is an issue where work is needed. But I am not sure
putting flags on dozens of functions is the best way (and a way for others to
understand the code).

Merging of code works fine until you follow these few regulars:
- do not confirm more than one work in one step. For every work use a
single confirm and give 
this a good explanation (not the date)
- don't delete or change settings or files you do not know where there are
good for. This can be settings files for eclipse or Linux Makefile.

I am working actually on implementing 128KB serial EEPROM instead of
virtual EEPROM. This is the first step: the next will be creating memory
places (100??) for storing concrete frequencies and station names.
Names can be set using USB-key. Routine for reading FAT from USB is
ready (I use this in my bootloader).

You can go on working on your pulled source - and you can go on pull
the "new cersion 219.24", too. So you can have a look on what I have changed.

But it is not the best way to dig complete code. When you look at github you can
compare version before and afte a commit. If the name of the commit is good
you can much faster understand changings.

Of course it is possible to revert a commit if it is problematic or faulty. It does not
care if the to be reverted commit is the last or not. All this works in absolutely most
of all thinkable situations.

vy 73
Andreas

cctbcn, 2015-10-13:
I made the fixes, but I'm not sure that anything actually happened as I can't seem to convince the App that I actually changed anything.  (e.g. "There aren't any commits to merge." so I can't send a pull request.)

I see "mchf_git_20151012-RC" online, but I can't see any files associated with it.

Rescue9, 2015-10-13:
If you made fixes you have to commit the fixes before a pull request will be submitted. If you are not seeing changers in your code, make sure you're on the correct branch. That was my issue a lot of times.

closed 2015-12-06 by df8oe