df8oe/UHSDR #13: Feature request:  Menu selection to force a specific TUNE power

cctbcn, 2015-10-30:
It has been requested that a menu item be added to (optionally) force the selection of a specific power level when the TUNE button is pressed.  The default would be "Off" (e.g. use currently-selected power level).

d9394, 2015-11-02:
1. And after the TUNE function , I suggest MCHF has a SWR scope scan function . 
2. I advise a new UI, resvre a blank line space to display more  infomation in the future.
[image]

cctbcn, 2015-11-02:
>     infomation in the future. noname

I'm not sure that I understand #1:  Is this function to be that of an 
antenna analyzer?  Surely this would be completely illegal considering 
the amount of output power (250-1000mW) required to get a useful reading 
from the reflectometer!

On #2 it would be helpful to provide information of an example as to 
what might be desired.  There is currently quite a bit of blank "unused" 
space on the display that I frequently use during debugging:  
Considering the amount of work required to do this it would be 
preferable to have a goal rather than just doing this for its own sake.

73,

Clint
KA7OEI

d9394, 2015-11-03:
Hi Clint,
    Yes, your means is right, that is likely an antenna analyzer, because MCHF is design as a portable mechine, so when I was using MCHF at field days , my antenna is temporary setting up, and then I needed an analyzer to find the best SWR frequence to be on air or I needed to check  my antenna being correct setup.
  I think the  SWR scan function is like  these: using a lower power ( it  can be setting  at menu item as a parameter) ,and SWR Scan Range is another parameter that can be setting at menu item ( such as 100KHz ), then start transmit at frequence : VFOA-Range/2 and end at frequence: VFOA+Range/2, then draw a SWR line in the waterfall area.

DG9BFC, 2016-02-29:
hello ghost

that idea is great ... and it is LEGAL 

what is difference to you turn the dial while transmitting .. or you turn in 5 or 10 kc steps .. measure ... and make a note on paper .. then tune next setting measure again

mchf could do that in a few seconds ... (making a note in ram does take as long as a cw dit)

ionosondes do that every day over complete shortwave bands ...

and if you use the mchf just for an antenna test on fieldday to prepare your setup ... it will be the same as you would do that by hand (making notes on paper) ... so it sure IS legal

(just do not use outside ham bands!! because outside you should not use more then a few milliwatts)

that is legal right???

that stepping could be done very fast ...

DG9BFC, 2016-02-29:
forgot to note 

tune NOW HAS THAT FUNCTION YOU ASKED FOR

(setting for tune power in menue ... so it is solved

df8oe, 2016-03-02:
mplemented.

vy 73
Andreas

closed 2016-03-02 by df8oe