df8oe/UHSDR #15: REQUEST: Color coded freq option

Rescue9, 2015-11-08:
As a new General Class holder, it would be nice to have two additional options in the menu.

1) Option to color code frequency based upon license class when frequency goes outside class privileges. {Off, Orange Yellow, etc.}
2) Option to select current license class.

This way, those of us who are new and are learning on out mcHF would have less chance to violate our c!ass privileges. I also believe this is mod necessary due to the portable nature of the radio and the small likelihood that one would have a band plan, or additional documentation, handy for reference.

Rescue9, 2015-11-09:
Issue in wrong repo. Closing.

df8oe, 2015-12-06:
Very good idea. It can be paired with known CW / RTTY WSPR and so on spans and be coloured differently. Also transmitting can be disabled if you leave the span where you do have license (overrideable). Table can be added via USB-memory-key in txt form. I will think about this...

vy 73
Andreas

closed 2018-01-30 by df8oe