df8oe/UHSDR #16: BUG 0.0.219.26.5

df8oe, 2015-11-12:
When toggling through modes and doing the step FM --> USB there is no more audio. Sometimes LCD slows and mchf is so slow you cannot push knobs. Only disconnecting power supply helps.

vy 73
Andreas

df8oe, 2015-11-12:
Hello Clint,

I have had a review of your commit 219.26.4 --> 219.26.5 There are some code cleanups which may be the reason for the bug above... Not the changing from "==" to "!="

vy 73
Andreas

cctbcn, 2015-11-13:
I switched to the newest GCC which spawned a slew of warnings and a few 
of which the compiler nearly demanded that I correct (e.g. I was 
initializing an array with more elements than it actually contained):  
It just goes to show that no good deed goes unpunished.

I couldn't replicate the problem on either of my UI boards, but having 
seen such problems before (e.g. some people having them, others not) I'm 
not terribly surprised that it does occur.

* * *

I am also trying to chase down a remnant bug - apparently new since 
after 219.22 - that *sometimes* appears when one tunes in 100 kHz steps 
and causes the Si570 to be loaded with bad data:  Only a power-cycle 
corrects this.

This appears to be in addition to the previously-mentioned DCO frequency 
range bug which manifested itself differently:   In that case, one could 
always find frequencies that *did* work.  I've seen it pop up before, 
but I thought it was a fluke - but it's happened on both UI boards with 
every FW version since the DCO range fix - including 219.24+DCO range fix.

I'm suspecting that, possibly, something should be a global/static but 
is not or, more likely, some unlikely set of conditions for which there 
are no valid "answers" in the tuning algorithm and it can never 
recover.  Again, it appears only rarely and only when tuning in 100 kHz 
steps as one might to tune from a ham band into an Shortwave Broadcast 
band.  When this occurs, the LO output appears in only a brief burst at 
some unknown frequency when the TUNE command is given, but otherwise 
there is no LO output at all, so it may be a defective tuning 
instruction for the DCO - but that's only a guess.

The Si570 tuning algorithm is one portion of the code that I have not 
reverse-engineered, so I only know vaguely its constituent parts, enough 
to have made (absolutely) necessary modifications to it in the much more 
distant past.

Clint

df8oe, 2015-11-13:
Hello Clint,

I am nearly sure new upcoming issues are related to cleaning up code. As I started viewing code half a year ago I noticed many compiler warnings that *seem* to remove easy. But it is not.... For example there are two array declarations (I mention "test a" and "test b" or so) which are "declared but never used". And warning is correct: so it is. You can comment out these declarations and warnings are gone - no errors. BUT: the binary does no longer work as expected.... I never had understand whats going on. Maybe there are more of such "interesting stories" in source. I think first you shall revert all code cleanings where you are not 100% sure they are harmless. If there is the smalles doubt: better leave it untouched. I can revert your changings via GitHub, too but this makes it harder for you to keep on sync. It will be easier when you create a copy of the existing testing, pull it to your local computer, do the changes, commit it, push it back to the GitHub clone. So it is a quite easy way for you to go. Of course we are running with "active handbrake" but better SECURE than digging into deep chaos...

vy 73
Andreas

df8oe, 2015-11-15:
Hello Clint,

I reverted your last commit to 219.26.5 because of it makes firmware unstable in some cases. I named testing noe 219.26.6 - but no further changes by me are included. New naming is only for better identification.

vy 73
Andreas

df8oe, 2015-12-06:
changes reverted - closed.

closed 2015-12-06 by df8oe