df8oe/UHSDR #43: CAT Support via USB Virtual Com Port

hogthrob, 2016-02-01:
I want to implement proper remote control of the mcHF using a virtual serial port.
The current devel-DF8OE will soon contain the base functions for doing this (a working virtual serial port with an interface to the application layer).

My current plan is to implement a simple, well known CAT protocol at least for the first time. This would allow to use the mcHF with existing software without much trouble. 

Since I know the FT817 protocol and is fairly simple, I would use this as a base. Implementation should allow to plug in other CAT emulation and of course at some point also a native protocol.

Choice of protocol could be handled in the config menu (reusing the existing CAT Mode Switch).

The purpose of this issue is to record the progress and to allow for these changes to be discussed.

hogthrob, 2016-02-02:
Since it seems that the current approach based on FT817 command set is working quite well, I think it is important to discuss the next step: How to activate CAT? 
I would think that CAT mode should be a permanent selection like most other configurations. If enabled, it does not eat much resources (almost none at all) if there is no connection to USB. Even if connected, it only waits until  5 bytes are received. Only then processing of the request takes some time.
So the overhead should be neglectable.

df8oe, 2016-02-05:
finished and working fine :) Many thanks for your work.

closed 2016-02-05 by df8oe