df8oe/UHSDR #49: Feature: Make LINEIN Channel Selection an configuration option.

hogthrob, 2016-02-03:
MCHF uses by default the left channel data for line in. Like in my case, the right channel would be the right channel. I would suggest to implement this as additional option in Menu 060 Mic/Line Select to be now MIC / LINE-L / LINE-R with MIC being the default. This would be 100% backward compatible but would solve my problem. 

I would add another ts element to define the used channel and use this in the appropriate places (audio_driver.c).

df8oe, 2016-02-04:
Thats absolutely ok. Feature does not afect anything and maybe helps others, too.

vy 73
Andreas

hogthrob, 2016-02-04:
Implementation is done, only issue I could imagine is the slight change in the UI interaction:
M3 long press now toggles through "MIC"-"LIN"-"L-R" since I opted to reuse the existing tx_audio_source parameter. That means you need two long press instead of one to get back to microphone. I don't know if this creates a usability issue. If so, it can be implemented with an additional  parameter which tells which line to use and you have to select the left/right in the config menu. With this switching between MIC and LINEIN and selecting the LINEIN channel would be separate choices. Both ways have its pros and cons.

closed 2016-02-04 by df8oe