df8oe/UHSDR #51: RF attenuator

IZ6MAF, 2016-02-04:
I'd like to equip mcHF of a resistive attenuator with relay controlled by a FET to be inserted instead of Q2 as in the attached diagram , the activation of which is controlled by a combination of existing buttons.
Please can reactivate the command line of the Q2 gate.
[image]

df8oe, 2016-02-04:
At this stage I won't do that. his line was unused for a long time and because touchscreen activation was done and needed a GPIO the only one I could use was this. A deactivation of touchscreen I won't do. But if you want I can add a configuration menu where you can choose whether to use this line for touchscreen or for attenuator. All others collide with functionality that is already present.

Do you have tried a much more linear prestage instead of using an attenuator? I have made great improvements by using a MMIC instead of a transistor. I use PGA103 and do not have any intermodulation so no need for an attenuator.

vy 73
Andreas

IZ6MAF, 2016-02-04:
If the PGA103 solves the problem of intermodulation I prefer it and there is no need of the attenuator .

Which of the two types PGA103 / PGA103+ you have used and from which provider you bought them?
Please could you send me a simple outline of how you have connected and powered (+5V or +3V)
Vy 73
Paolo

A: df8oe/mchf-github
Cc: IZ6MAF
Oggetto: Re: [mchf-github] RF attenuator (#51)

df8oe, 2016-02-04:
I have use PGA-103+ from ebay (United States seller). I have powered via 5V.

There is no schematic available. The steps are easy:

Remove R35-R39, C66 and Q1.

Solder in PGA103 which only needs three connections:
output to T1(4)
GND
input to C64

Then you have to adjust R35 which is in range of 4.7R...39R. Because of current you need a resistor with higher wattage - I used a wired resistor for testing and when I found best value replaced it with 1205 SMD.

Testing is only for IM breakdown - amplification is big enough whole voltage range.

vy 73
Andreas

IZ6MAF, 2016-02-04:
Thank you very much Andreas , I order immediately the PGA103 +

vy 73 Paul

df8oe, 2016-02-04:
Hello Paolo,

please *do not quote* previous emails. They are correctly sorted to relevant issue on Github - there is no need for quoting (like on Yahoo-NG :/ )

This mod is not completely and well tested so I do not have published it yet. It would be great if you can give a feedback after testing!

vy 73
Andreas

IZ6MAF, 2016-02-04:
Ok Andreas after testing I will give you a feedback by email

S52CQ, 2016-02-28:
Hi guys, I was also thinking to make an RF attenuator, as I live close to local Broadcast station with 350 kW in antenna I need it. 
Without change of SW I did make a flip-flop circuit that is triggered with a switch on one of the encoders. Flip-flop drives relay which turns attenuator on/off. For attenuator I have used a circuit with 2 resistors - same as Kenwood TS820 has. The proof of concept worked nicely, soon I will design a small SMD PCB with flip-flop, relay and resistors.

73 de Jure, S52CQ

df8oe, 2016-03-02:
Hello Paolo,

have you tested MMIC instead of transistor?

I'll close this because there are several solutions without spending one more GPIO. Maybe we do have much more than in future because LCD SPI mod works excellent now and there are some larger LCDs which are SPI-compatible with HY28 ;)

vy 73
Andreas

closed 2016-03-02 by df8oe

IZ6MAF, 2016-03-02:
Hello Andreas 

Before I put in parallel with C65 a tantalum capacitor 1 uF, then I have tested three different solutions with PGA 103+ instead of Q1

1)      Power supply + 5V from C67 by 34 ohm resistor (68//68 ohm) 

2)      Power supply + 5V from C67 by 9,4 ohm resistor (4,7+4,7 ohm) 

3)      Power supply + 3,3 V from C37 by 4,7 ohm resistor

In all three cases the gain seemed to be excessive and the intermodulation in the evening on 40m were worse that with the transistor.

I also tried to reduce the gain by installing a 50->100 ohm resistor between the signal output of the PGA103 and ground, getting a result very similar to that with the transistor.

At that point I reassembled the whole circuit with the transistor; pity that I had bought five PGA 103+ to replace the Q1 transistors on our equipment !!

I'd be curious to know which configuration you're using for the PGA103+ on your mcHF, and what results you got.

Vy ’73 Paolo IZ6MAF