df8oe/UHSDR #65: Request for Contributing Guidelines

db4ple, 2016-02-15:
It seems, the time is right to start putting a Contributing.md together. I propose to put a first rough draft into the repo, discuss and extend/revise first using this issue. And do it quickly. Once we have a good initial one, let us then handle change requests as any other contribution following the procedure outlined in the Contributing.md :-)

df8oe, 2016-02-15:
I agree 100%. Community is growing and that is very positive. At this stage such a guide will improve our working together significantly.

Many thanks!

vy 73
Andreas

db4ple, 2016-02-15:
Some samples:
Good structure, not too long:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/contributing.md (Attention, copyrighted!)

Very short:
https://github.com/thoughtbot/factory_girl_rails/blob/master/CONTRIBUTING.md

df8oe, 2016-02-15:
# Contribution Guide

This document describes some points about contribution process for mcHF firmnware and bootloader.

The maintainer of this GitHub is DF8OE.

The project is free for non-commercial use and is being developed within community. Maintainer merges pull-requests, fixes critical bugs and manages releases. Participation in work is appreciated.

**Pull-requests**

If you fixed or added something useful to the project, you can send pull-request. It will be reviewed by coders and/or maintainer and accepted, or commented for rework, or declined.

We love pull requests. Here's a quick guide:

- Make sure you have a GitHub account
- Fork it
- Create your feature branch (git checkout -b my-new-feature)
- Commit your changes (git commit -am 'Add some feature')
- Push to the branch (git push origin my-new-feature)
- Create new Pull Request
- Please make sure you add a test for your change and all tests are passed. (bundle && rake test)

**Bugs**

If you found an error, mistype or any other flawback in the project, please report about it using issues. The more details you provide, the easier it can be reproduced and the faster it can be fixed. 

**Features**

It you've got an idea about a new feature, it's most likely that you have do implement it on your own. If you cannot implement the feature, but it is very important, you can add a task in issues and tag it with "REQUEST:". Feature requests are discussed and may be realized shortly, later or never - there is no guarantee that requests will be accepted.

db4ple, 2016-02-15:
Tests: Do we have an approach for the tests? I am not aware of that (but I think that is important)

Coding Guidelines: I also would like to have some coding style guide lines, not too much but some.

Features: I believe that for any "larger" feature even if you do it on your own there creation of an issue should encouraged, together with a statement of who is going to implement it and when (if known). This would help to coordinate things and avoid rework / frustration if pull request can or will not be integrated.

df8oe, 2016-02-15:
Tests: We do not have but we need to have...

Coding Guidelines: Yes - would be nice

Features: collecting ideas and managing hould/can be made simply via issues discussion.

New idea: WIKI should come up and contain operating manuals, modification lists and so on. Should also be a part of the guide?!

vy 73
Andreas

db4ple, 2016-02-15:
I refined my previous entry. 

Wiki: yes, for "slowly" changing content and things more related to mcHF as a device as opposed to the SW development.

db4ple, 2016-02-22:
Andreas,

could you place an initial CONTRIBUTING.md file in the project? I think this way it is easier to refine it. As not many joined the discussion here, I think we should close this and go on with modifications via pull request.

df8oe, 2016-02-22:
I have placed the above one on each branch.

vy 73
Andreas

closed 2016-02-22 by df8oe