df8oe/UHSDR #67: New Feature: USB Audio Support

db4ple, 2016-02-16:
I am currently working on getting the USB AUDIO support going. Goal is to provide audio in and audio out via the standard usb. At first this will not be coupled with CAT to keep CAT mode working as is. Once stable operation of audio channels are achieved, both will be combined, giving you (and me) the ability to run digital mode QSOs via a single thing USB cable. 

If possible also an IQ in/out mode going to be implemented. Shouldn't be more difficult than a mono audio stream but who knows.

The first commit will provide a digital microphone with a simulated (!) signal.

To enable this you have  recompile with USB_AUDIO_SUPPORT defined in mchf_board.h and to enable CAT mode (yes, CAT mode). But for this DF8OE has to accept my pull request :-)

db4ple, 2016-02-17:
I made some progress. AUDIO IN worked quite well it seems. However, AUDIO OUT needs some testing.

And I was not able to combine the descriptors into one that was able to operate both interfaces together. Linux recognized it but not sound could be received. Needs investigation.

db4ple, 2016-02-21:
Open Items:

- WSJTX Linux: Cannot see mcHF Input in list of available capture devices
- I/Q RX not possible, only TX
- USB Strings need some tending
- Maybe some gain control should be added (right now fixed, displayed settings are not used as these are for line gain)

df8oe, 2016-02-21:
Regarding #1:
This is related to audio management - not to mcHF. I can work using wsjtx using pulse audio as mixer.

DG9BFC, 2016-02-21:
usb audio works here (more or less)
but
does not give I/Q data (i see both sidebands!!) but processed audio 
and it should not use 0hz out but 12 kc

i used the I/Q data from 30 pin connector (routed to my onboard soundcard) and can see the same as on the mchf display in hdsdr ... and can sync hdsdr wirth mchf via cat in both directions

(hdsdr set to receive 12 kc I/Q data)

also receiving of drm does work (soft sodira and dream used for testing)

but that only works with RAW I/Q data and not with filtered data

bw seen in hdsdr depends on setting of am filter width (so it is processed audio)

my idea is ... when setting digital .. then normal audio is given in and out

(line in and line out via usb ... processed audio!)

when setting digital I/Q ... then it should be RAW data in and out (12 kc I/Q data if you set plus or munus 12 kc in menue) ... the 0hz spot should be avoided when using I/Q in and out (as it should be in any sdr and why we have the plus or minus 6/12 kc setting)

then i should see similar bw in hdsdr and can make the gain and phase correction in sdr soft

and the signal that the (simulated) soundcard gives is WAY toooo high and i can not turn it down

long press of the button below enc 3 and i can select dig or di I/Q ... but turning encoder does nothing (is that for I/Qi input like mike gain??)

there should be a setting of digital I/Q level OUT towards pc (is way too high now) and a setting for I/Q in level ...

in windows I can only see the audio out mixer but not audio in  ... so no way to turn the volume down (signal input level towards hdsdr or other soft)

so ... in menue ... select (processed) audio or (raw) I/Q with setting to digital or digital I/Q

on the front panel with long press of that button below enc3 select DII for dig input and select DIO for output level/gain ... so that you can tune both settings for best recieve (or needed tx level)

(that has to be audio level or I/Q level depending in the setting of digital or digital I/Q in menue)

db4ple, 2016-02-21:
Many items on your list:
- Yes, as stated I/Q RX is not yet supported. 
- I am not sure about frequency shifting in the mcHF. I'd rather not want to process the raw I/Q data and leave the heavy lifting to the connected PC due to increased load. But of course it may not require much resources on the mcHF. I don't know that. 
Let us get the DIQ working and see how  to proceed.

- Yes, current plan is that when DIQ is selected, output switches to I/Q on digital.
- Digital gain control for the processed audio sent to USB is fairly easy to implement, it just needs 
  a setting separate from the mic/line gain.   For DIG IQ see above. 
- On Windows you can control the input level using the "Level Properties" section of the STM Input (Recording Devices), I set level 2 there.

DG9BFC, 2016-02-21:
Hi danilo

> Many items on your list:

Just two ... grin

> 
> *	Yes, as stated I/Q RX is not yet supported.

> *	I am not sure about frequency shifting in the mcHF. I'd rather not
> want to process the raw I/Q data and leave the heavy lifting to the
> connected PC due to increased load. 

Pc easy can process that load and for working with sdr soft you need that
RAW data in the pc

I know that you then have to send MORE data via usb ... (the complete
baseband signal that the sampling chip sends to mcu)

But only then you can use that raw data in sdr soft (or to receive drm with
sodira or dream)

>But of course it may not require much
> resources on the mcHF. I don't know that.
> 	Let us get the DIQ working and see how to proceed.

To have it properly working you have to make two signal paths selectable

One selected when set digital and the other selected when set to digital I/Q

One should be processed audio (taken behind the processing)

And the other should be raw I/Q data (taken before any processing)

Then we would have

Mic in, lin in (l+r), line in (via usb), I/Q in (via usb)

> 
> *	Yes, current plan is that when DIQ is selected, output switches to
> I/Q on digital.

Input and output ... BOTH IN RAW DATA!!! Means if you select raw I/Q  data
in and out and 12 kc shift in menue then also 12 kc if signal is sent and
received via usb 

That would mean you have a non processing frontend and the pc has to do all
the work

> 
> *	Digital gain control for the processed audio sent to USB is fairly
> easy to implement, it just needs a setting separate from the mic/line
> gain. For DIG IQ see above.

On front select d i i for in and d i o for out

You already have those two selections ....buttons ... turn knobs

> *	On Windows you can control the input level using the "Level
> Properties" section of the STM Input (Recording Devices), I set level 2
> there.

I only can see the output mixer but input is not shown ... do you know how
to add the input mixer in my old windoze system?? (running xp here and used
the win7 driver)

In sdrsoft the input is overdriven (20 db above 0db line) ...

But that is audio not raw data ... I will see what I get if you changed the
signal line (sent towards pc) from audio to raw data (baseband output)

And then have to see how to add that input mixer

Greetz

Sigi

Dg9bfc

Ps ... we can also discuss in german if that is easier for you

db4ple, 2016-02-21:
I just added digital IQ as promised. See pull request #77 , Andreas will surely merge that quite quickly.
Tried it with HDSDR. Support for XP, well. Can't do much for this system as I don't have one. However, some changes to the USB Audio descriptors (and the code) will most likely have to happen since also on Linux without pulse there is the Input missing for whatever reason. Probably fixes the XP problem as well.

German may be easier for me. But not for most github users.

DG9BFC, 2016-02-21:
> -----Urspr�ngliche Nachricht-----

Baseband out works superb!!!!!

I can now switch between my onboard soundcard and the stm32 soundcard and
get the same input signals on both ways

(did not try out drm decode or other things ... just checked with that
sdrsoftware)

WELL DONE :-)

Dg9bfc

Sigi

Ps ... will tell my other results after further testing

DG9BFC, 2016-02-21:
> -----Ursprüngliche Nachricht-----

OK ... baseband out tested with drm decode

Works superb!!!!!!!!!

Voice of Nigeria received up to 30db s/n (most time 20-24 db)

Noisefree audio

Baseband out via usb gives same signal as the onboard card connected to 30
pin connector (raw baseband signal)

Tested with hdsdr (all analogue modes) and with sodira (decode drm)

.............................

WELL DONE!!!

(only thing that is missing is a level control for that signal

Digital in and out)

The settings on front are for input only????

I see DIG and DIQ on front

Could you change that to: 

D I I ... AND D I O

(digital in and digital out)

That settings then could lower input and output level ... for audio (if set
to digital in menue)

Or for baseband (if set to digital IQ in menue)

Then we would have

Mic in

Line in left or right

Line in and out via usb (digital)

IQ in and out via usb (digital IQ)

(we already have that but the level control in front is missing and that
level control could change analogue audio or baseband level)

Dg9bfc

Sigi

db4ple, 2016-03-04:
Andreas, if testing is okay, I would suggest to close the issue. I did all I wanted to do in the initial round.

DG9BFC, 2016-03-05:
Usb audio works ... cat works

Iq data works

Yepp .. I think all is solved :-) (if it works on ALL windoze)

Dg9bfc

Sigi

df8oe, 2016-03-05:
successfully finished

vy 73
Andreas

db4ple, 2016-03-05:
Then it is closed. For now.

closed 2016-03-05 by db4ple

Jmelz, 2017-09-15:
I seemed to have found an issue - or maybe my mchf has a problem - but when I transmit a tuning tone with any app from the windows 10 pc, I get a cyclic static burst (about a fraction of a second, every 3 seconds). When outputting audio from the headphone jack of the pc to the line in ports of the mchf, the audio is clear. Does not matter what band/frequency, and I can easily repeat this behavior.