df8oe/UHSDR #100: Project licence

jkramarz, 2016-02-27:
One more think for today: I'm wondering about the exact meaning of "For radio amateurs experimentation, non-commercial use only!" licence.  If it's "experimentation only", it may exclude regular chit-chat usage of mcHF, as I won't consider talking to the same friend from time to time an experimentation. The same would affect my primary intended use-case of this radio - I'm interested in activating some climbing-only summits in SOTA program ;-)

What do you think about making it CC BY-NC-SA 3.0?

- it's a quite good license with proper legal text, translated to most of languages,
- I'd consider "non-commercial" and "for radio amateurs" redundant, as it seems there's no legal way to use it outside amateur bands - the other ones are licensed and intended for commercial use

So it seems to be compatible, but well defined.

jkramarz, 2016-02-28:
> @df8oe:
> I cannot change license because the project has already license and its:
> https://creativecommons.org/licenses/by-nc-sa/3.0/
> The text in splashscreen is grown and only a "hello" to users - not a license description.

OK, so if it's already CC BY-NC-SA 3.0, can I update licence description in source files?

df8oe, 2016-02-28:
Of course you can do this!

vy 73
Andreas

closed 2016-02-28 by df8oe