df8oe/UHSDR #99: Dual use of PA9

jkramarz, 2016-02-27:
Hi there,

I'm wondering about the use of PA9 pin, as it is UART1 TX for debug builds and TP_CS at the same time. As ST-Link/V2 got proper support in OpenOCD, what do you think about switching from UART to Semihosting for debug purposes?

Best regards,
Jakub

jkramarz, 2016-02-27:
More info regarding semihosting:

- http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0471g/Bgbjjgij.html
- http://www.wolinlabs.com/blog/stm32f4.semihosting.html
- http://wiki.dlang.org/Minimal_semihosted_ARM_Cortex-M_%22Hello_World%22

It's quite easy to setup.

db4ple, 2016-02-28:
Old debug printf calls should be removed :+1: 
Semihosting availability in the project and instructions how to use :+1: 
Need for semihosting: Don't think so for the moment, as semihosting also influences the timing. We are better of with direct inspection for most parts (and you can always write to the screen, if you  want). Although for USB stack debugging, it would be very nice to have this.

closed 2016-02-28 by df8oe