df8oe/UHSDR #110: Encoder

S52CQ, 2016-02-28:
Hi,

I have noticed in the dev. ver.0.219.27.4 a problem with encoder. It does happen quite often that encoder does not follow rotation of the main Freq. button on 100 Hz an 1 kHz setting. I did not notice that also at Volume button. Very seldom I could see freq. number displayed in Red colour if that does make sense.

There is no rule, when this happens. I may leave encoder alone and start turning the button and I can notice that encoder is not following, or I might quickly rotate Freq. button left right and suddenly I can notice that encoder is not following. Also I did not notice if this happens at low or high receive signals.

73 de Jure, S52CQ

df8oe, 2016-02-28:
I can confirm this but I have thought not-precise encoders are the reason. It occurs sometimes and not depending on anything. We will have this in focus.

S52CQ, 2016-02-28:
Hi,

I don't mean to offend anyone, but I did not notice that in regular versions - if this helps anyhow? This has started to happen at ver. .27.4. 

73 de Jure, S52CQ

df8oe, 2016-02-28:
I have had unprecise reactions of encoders n from the beginning so I have not recognized there any change. But we will have this in focus.

vy 73
Andreas

DG9BFC, 2016-02-29:
what i found out is that as more as the mcu has to do (example notch or dsp if re-activated in menue) ... as longer it takes to react on a button press or a dial turn (regardless be it vfo or volume or other things)  

also waterfall speed setting and other setiings in menue ... or if zoomed or full view .. all that sets the speed of the mcu and so can slow down the reaction on any button press or turn

.... and the dsp settings (especially noise reduction!) can slow down the mcu so much that it can be totally unusable (even power off does not work anymore!!) ... that is the reason why nr is off by default in menue ... it can bring the mchf to sleeping (yes it still works ... but you can not change any setting or switch it off and if set tooo high it can switch off completely!!)

that is normal

remember the mchf is nothing more then a pocket calculator with a soundcard and an rf mixer with pa .. so reaction is slow if mcu has much work ... sdr is very cpu intensive (try pc sdr software on an older machine and you see what i mean)

and now ONLY FOR HIGH RISKY EXPERIMENTERS :....(be aware that you can make mcu sleep with too high settings!!)

all comment below only are for WIDE AM FILTER AND NOTCH+NR ACTIVATED (5 kc or more ... 3.6 works normal) swo for that test you have to reactivate nr (it is not stored and so off in next power up)

and it is only when you activate BOTH (nr and notch) AND you select that wideAM filter

(reason why the notch+nr is normally not selectable in wide am and you have first to activate notch+nr with a smaller filter and then select am wide)

AND WHY NR IS DE-ACTIVATED IN MENUE (go figure)
.
........

you can crank up all dsp setting to max with ONE EXCEPTION!!! (i found out with try and error but as already noted ... tooo high settings make the mcu nearly sleeping!!!!  and it is a risky test!!!)

menue 315 DSP FFT NUM TAP is the point that makes the mcu sleep!!!

and now a list (thats what i found out here with try and error ... RISKYYYYYYY!!)

set at 64 or below all is fine (mcu runs fast even with notch and nr activated) ... with am wide and nr+notch it is slow ... but usable

set to 80 ... mchf still works but is further slowed down (in my view a bit too slow but it still works!)

set to 96 ... (the default setting!!!) ... if you use nr+notch+am wide it is VERY VERY SLOW ... not usable in my view  (but can be switched off w. power button or with longer press or filter button switch off dsp .. ) reaction is slowed down to almoszt zero ... regardless how fast you turn vfo ... reaction is slowed down to nearly fallen asleep

the absolute maximum!!!!!!!!!!!!!!! but if you ask me ... i would say it is tooo high and so DO NOT USE THE DEFAULT!!

if you set the value to 112 or 128 ... no way ... mcu is fallen to sleep .. or can be totally dead .. or suddenly switch off or ... you gave the mcu so much work that it can not react on anything ... not even power off does work ... nor can you switch off dsp with the long button press ...

MCU FALLEN ASLEEP OR SWITCHED OFF (or coocked ?!? just joking)

if it is not switched off cause mcu is totally fallen asleep .. power off only works with PULL THE PLUG

DO NEVER EVER USE SUCH HIGH SETTING IN NUM TAP!!!

I NOW HAVE MENUE 315 set to 64 ... (setting i would recommend as default!! ... 112 and 128 should be deleted from menue!!!! maybe also 96 ... so 80 would be max setting)

when you listen to a clear am station ... no nr needed and the use 5kc filter is fine

if you have some noise and need nr ... then use smaller filter

but if you accidently switch on nr + notch plus wide am ... you still can use (a very slowed down) mchf

S52CQ, 2016-02-29:
Hi,

I am testing latest version that I have uploaded just now. Encoder is now running much better, in fact I did not notice problems that I've seen before.
I did however find a small problem: in the 335 menu I did notice that I have to turn encoder to the left (instead to the right) to increase the number of Tune power level.

73 de Jure

df8oe, 2016-03-02:
OK - closed. Turning direction is swapped next time.

Many thanks for your report!

vy 73
Andreas

closed 2016-03-02 by df8oe