df8oe/UHSDR #114: antenna analyzer function (swr sweeps in coarse or fine steps)

DG9BFC, 2016-02-29:
i grabbed the idea and make a new issue here

add a function where the rig can step in coarse steps across a band (or several if not all bands) and on every step sends a short cw note (a dit should be enough)

mchf notes the swr and draws an swr sweep (or shows point of lowest swr or ...)

only inside ham bands ... and lowest power where we have a good readout (with all needed bridge modifications that has been found out)

that should be legal cause it would be similar as you would do that by hand (step in 10 kc steps... press a dit and make a note on paper) ... but it would be much faster

maybe stepping could be made coarse for bigger sweeps (and wideband antennas) ... and a bit denser for narrowbanded antennas (say a mobile whip for 40 or 80 or a mag loop)

guessing that 5, 10 and 25 steps are ok (depending on what antenna you test)

df8oe, 2016-02-29:
Because of the SWR detection circuit design which is used in mcHF you need minimum of 5W TX power to get evaluable results. Due to design of TX you must transmit much longer than a simple DIT or DAH to get a result which can be used.

Antenna analyzers do use amplifiers in their detection circuits which make it possible to work with *very low* TX power - that is quite legal and does not hurt my understanding of HAM-spirit, too.

Sweeping with TX power of 5W and transmit durations of 1...2 s over a wider range without listening before what is going on there MAY BE LEGAL - but this does hurt my understanding of HAM-spirit! When you answer "only one signal for a quite short time" - what about if more OMs are acting this way? And if you answer "but they don't do" you place your own private claims :/ We complain about growing QRM and act in such a way??

So I suggest the usage of a low-power antenna analyzer which is available for < 70,-- Euros will be much better than "misusing" mcHF for this taks.

I by myself do not want to add such a feature to firmware!

vy 73
Andreas

DG9BFC, 2016-02-29:
> -----Urspr�ngliche Nachricht-----
> or fine steps) (#114)

Ok .. sorry about that ... i was not aware that such high power was needed

......

I have thought that 0.1 w or lower would work (with the bridge mods) ..

With 5w you can work the world ... and yepp antenna test with 5w ... that is
NOT HAM SPIRIT LIKE

So forget about all that

Greetz

Sigi

Dg9bfc

df8oe, 2016-02-29:
Detection is made with schottky-diodes and output voltage does not raise above 3.3V at 10W (I have tested with 25W and voltage is around 3.3V). So its a mathematical game what minimum voltage must be ;)

vy 73
Andreas

closed 2016-02-29 by df8oe

DG9BFC, 2016-02-29:
7v at roughly 1w?!?

P = u x u /r

 7x7 ... = 49 divided by 50 .. abit more then 1w ... but that is output
voltage direct onm antenna connector ...

That does not include the bridge factor

Could you measure (just for info not for adding that function) what voltage
we have behind the bridge with say 0.1w or with 1w??

No I do not wanna press you towards the analyzer function (but it would be
nice ... right?!?) .. i just wanted to know from what values we are talking
about

Any user can use the mchf and send a short signal to check his antenna ...
no need for analyzer function ... but I liked the idea :-)

Greetz

Sigi

Dg9bfc