df8oe/UHSDR #117: Display error in 27.5 (centre line two times??)

DG9BFC, 2016-03-01:
tested new version (27.5) ... i have set off set to +12 kc (means i receive on left side of display)

when i receive on 7. 000 .. there is ANOTHER centre line (i have set to green) at 8 kc above the centre line (so can be seen at 7008 ... so one of the lines that should be red (my setting for the other lines) is instead in green ...

.................................

centre line always visible...>>>>

and I wanna ask if it would be possible to draw the centre line to bottom (like it would be with no signal) and the spectrum should NOT draw over it ... means centre line ALWAYS in green (or what user has set)

all other (in my settings red) are ok to be drawn over ... but i would prefer the centre line to be always visible in full height ... (yes ... a narrow cw note would be hidden behind ... is that the reason why not drawn to bottom)

df8oe, 2016-03-01:
I have tested and I can only find:
If you are activating scope you have a very short line starting from the bottom of spectrum which is mostly overlayed with spectrum but sometimes in a very clear moment its visible - no issue in waterfall??

vy 73
Andreas

db4ple, 2016-03-02:
Recently I had a look at this code and it needs some thinking. I also did some changes in the spectrum scope routines, which were not supposed to change anything. Maybe they did change something.

That the line is not visible in Waterfall comes from the fact that no one cares for this line once it has been drawn initially. 
Simplest solution would be to redraw the vertical line by the end. Given the time it takes to do the waterfall in general, this extra line doesn't add much more time.

DG9BFC, 2016-03-02:
> -----Urspr�ngliche Nachricht-----

The double centre line has to be deleted (that is a fault!)

Can not be seen in zoomed view ... just in 1x view ... and only in spectrum

It is like a max peak hold function (it reacts so)

When there is no signal then that second (green) line is nearly invisible
(hidden behind noise floor)

If you turn the dial that a signal reaches that spot then the green line
gets longer (higher) ... and stays there

Dg9bfc

Sigi

df8oe, 2016-03-02:
fixed.

closed 2016-03-02 by df8oe

DG9BFC, 2016-03-02:
Tnx ... issue can be closed <:-)

....

Also tested dynaic tune speed ... works superb

When set to am 5 kc stepping and turning (very) fast ... then speed goes up
to 500 kc stepping ... that way we can (almost) delete the band plus minus
buttons :-)

If you turn fast you are in a few seconds from one band to the next

(no ... I will not ask to delete those buttons ... grin)

Greetz

Sigi

Dg9bfc