df8oe/UHSDR #130: CW, LCD, 2.3k filter

S52CQ, 2016-03-05:
Hi,

I have one of the test versions (I wish that in the Blue menu there was Ver. No. shown) and I am testing CW. It appears that at CW mode the Monitor level (STG) is not displayed as it was - right to Volume (AFG). Now there is CMP, which has nothing to do in the CW.

Also I noted that at CW speed (WPM) near right digit there are two little dots, that can be seen at WPM set to to any number.

I presume that 2k3 filter remains ON all the time regardless setting in the Menu. Could it be turned off in the CW mode perhaps?

73 de Jure, S52CQ

df8oe, 2016-03-05:
Hello Jure,

you do not have the test vesion - you do have a "devel-version" :)

In the devel it is impossible to show version numbers. Sometimes there are 10 or more new versions a day - devel is "unstructured" regarding this point.

Regarding your issues you are all right. We will fix that in the next devels - I will start just this moment. Many thanks for the feedback!

vy 73
Andreas

df8oe, 2016-03-05:
STG - fixed.
WPM - fixed.

Filters need a little bit more work - I think finished tomorrow.

vy 73
Andreas

df8oe, 2016-03-05:
filter - fixed.

Happy weekend - vy 73 Andreas - closed.

S52CQ, 2016-03-05:
Hi Andreas,

that was fast - can I ask you a favour please? I do not have (yet) set the Eclipse dev. environment, can you please send me to my email compiled last dev. version please? Email is: jure04@svet-el.si

Thanks Andreas!
73 de Jure, S52CQ

df8oe, 2016-03-05:
Hello Jure,

no need to do this "only for you" :)

You can get binaries of devel-Branch as "daily snapshots" from our German project page:

http://www.amateurfunk-sulingen.de/mchf-projekt/modifikationen#start

Links for this are at bottom of the page.

I do not offer this at the moment n Yahoo group because philosophy is different. German group *loves* expermintal and new functions - Yahoo group does not.

Be free to download binaries there!

vy73
Andreas

S52CQ, 2016-03-05:
Hi Andreas,

tnx. for the link, that's fine too.

73 de Jure,.S52CQ

S52CQ, 2016-03-06:
Hi Andreas,

I am playing and testing SW the last one: mchf_06_03_2016_172723..bin. I hope that you don't mind me "throwing at you" all the issues. It is all with the good cause.

So the 2.3 kHz Filter is OK at the CW now. When I turn it Off it stays Off at CW. It is strange however, that when I turn on 1.8 kHz Filter in the CW mode - and turn ON menu #028, nothing happens, filter 1.8 kHz is off.

I have seen that at USB 1.8 kHz filter has written 1.8 kz, not kHz, same goes for 2.3 kHz filter, while at 10 kHz filter its written 10 kkz instead of 10 kHz. 

I did notice that at USB I have turned on only 1.8 kHz filter and it appears that I can select 2.3 kHz, although it's turned off in the menu 506. 

Filter Menus in the blue screen appears that do not erase OFF/ON properly. OFF is displayed Ok while ON is displayed as OON. That goes from Menu 513 onwards. Probably these filters are not written yet...

I did see problems with Encoder at CW&SSB again. 

And finally at filters setup, Menu 501 has written 500 HZ instead of 500 Hz.

73 de Jure, S52CQ

db4ple, 2016-03-06:
Lots of changes have happend on the filter front over the weekend by DD4WH and me. Not all is finished yet.

There is a table describing more or less completely how the filter selection works:
http://www.amateurfunk-sulingen.de/forum/attachments/Modes_und_Filter_fuer_Danilo.ods

And have a look at audio_filter.c, it should be almost selfexplanatory, there is a filter table
and the algorithm  to select the next filter with lots of comments. 

Have fun,
 Danilo

df8oe, 2016-03-08:
We are working on the manu and the configuration. Works will be finished end of the week. These issues are gone then - I close this now. But new version needs intensive testing again and then bugfixing of such things will start again

vy 73
Andreas

closed 2016-03-08 by df8oe