df8oe/UHSDR #138: Dimming Level and TX Inhibit

HB9LFU, 2016-03-07:
Hello mcHF Community,

The displays used in the mcHF are not all the same as you may have noticed.So sometimes I use not the full brightness and would like to have the dimming level integrated into the menu instead of always having to push several times on the power button. 
And it would be a good idea to use this power button function now free for a one-push display-off function. 

It would also be helpful to offer two different versions of TX Inhibit. One absolute with no chance of enabling it again without using the menu and one which can be changed by pushing the TUNE button for a longer time. 

Roger
hb9lfu

db4ple, 2016-03-07:
Hello Roger,
thanks for these requests. May kindly ask you to split these into 2 issues, so that these can be followed up individually. Thank you!

db4ple, 2016-03-07:
For the second part: One push of two buttons: POWER+ BAND- will do the job.

HB9LFU, 2016-03-07:
That is fine and not too complicated indeed! Thanks!

Regards
Roger

S52CQ, 2016-03-08:
to continue LCD debate... When working portable it would be handy that LCD is off all the time when freq. once set. Namely as it is now, every change (RX/TX or change of Volume) turns LCD on, but as mentioned when portable every mAh counts. Just a thought...

73 de Jure, S52CQ

df8oe, 2016-03-09:
@Jure
We will have many users who do not agree to this - I am sure. A quite good solution for users who wants to fly their mcHF in "dark mode" is setting time to 1 sec. That are not too much mAh which re wasted...

vy 73
Andreas

df8oe, 2016-03-09:
@Roger.

Normally I get questions exactly the other way. Because of dimming is done via PWM receiving gets oisy when LCD is in dim mode. So most others want to use LCD dim rarely and when activated that it is OFF next power on....

Really create a configuration for this??

vy 73
Andreas

HB9LFU, 2016-03-09:
Hello Andreas,
 when everybody can choose in the menu which dim level he/she desires, even **off**, it is fine. 
Those who hate the additional pwm noise can be satisfied with this as well.

df8oe, 2016-03-09:
At the present state configuration menu is already too big and you must read many screens before you reach the point you want to go to. So at this time no configurations will be added.

We are working ona completely different configuration meno which is splitted to technicalsectorsand in each sector you can find related items. If we reach that stage we can reopen the discussion again.

vy 73
Andreas

HB9LFU, 2016-03-09:
Hello Andreas,
yes, that's very true. There should be subdivisions in the menu to help find the entry desired easily. At present it is rather an effort.

jure, 2016-03-09:
I probably shouldn't be included in this discussion? I think you mistook me for @S52CQ.

df8oe, 2016-03-13:
Sorry Jure - my mistake :)
vy 73
Andreas

closed 2016-03-13 by df8oe