df8oe/UHSDR #139: Output

oz8gw, 2016-03-07:
I have replaced the driver transistor DXT3150 with the BFQ18A. Now I cannot turn down enough to low output on the lower bands. Is it possible to lower the output more in the menu.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-03-08:
The possibility of these settings will be adjusted to that. I noticed that, too. But already using the "old" transistors setting must be in the first quarter so its level is generally too high.

vy 73
Andreas

oz8gw, 2016-03-08:
Hi Andread.
Is it here to report about bugs.
Vy 73 de OZ8GW, Leif.

Den 08-03-2016 kl. 08:49 skrev DF8OE:

df8oe, 2016-03-08:
Hello Leif,

yes - and I agreed that it is a bug and that it is "under work".

vy 73
Andreas

df8oe, 2016-03-16:
When working on bands 40m or lower internal main TX gain is reduced 6dB. This should fit driver transistors BFQ18 / BFQ19 / DXT3150

vy 73
Andreas

closed 2016-03-16 by df8oe

df8oe, 2016-03-18:
Hello Leif,

many users of Yahoo group do not believe that power increases so much when changing drivers that it can't be set to recommended levals at lower bands. There is a discussion on Yahoo for reverting tis firmware mod. Would you please be so kind and write your output adjustment behaviour before / after change of gain in firmware with your mcHF on Yahoo-NG?

vy 73
Andreas