df8oe/UHSDR #140: TX Inhibit

HB9LFU, 2016-03-07:
Hello mcHF community,

It would be helpful to offer two different versions of TX Inhibit. One absolute with no chance of enabling it again without using the menu and one which can be changed by pushing the TUNE button for a longer time. 

vy73s
Roger

df8oe, 2016-03-09:
Please explain what is the advantage od this method - I don't see it...

vy 73
Andreas

HB9LFU, 2016-03-09:
Hello Andreas,
the advantage is that this way one cannot accidentally activate the transmitter without using the menu. Or have you ever thought of selling a mcHF to a short wave listener. I refrain from calling for a password protection but it should at least not be possible to transmit again by fumbling around with tune button when you have beforehand switched the menu to tx inhibit "on".

df8oe, 2016-03-09:
It is not possible to prevent anyone from transmitting. A SWL never touches TUNE button because he does not need it. So it nearly impossible to activate transmitter by chance. At the other side a SWL of course must use menu and configuration menu so he can activate trnamitter by chance there, too.

I do not see the advantage.

vy 73
Andreas

HB9LFU, 2016-03-09:
Hello Andreas,
and I find a menu point where you expressly state that you want to inhibit tx ridiculous when afterwards you can bypass this by using the tune knob. There is no such discipline or human perfection that we never inadvertently push the wrong buttons or push the correct buttons a second too long.

df8oe, 2016-03-13:
Added as new feature to devel branch.

Setting in CONFIG now overrides setting by pressing TUNE button long. If TRANSMIT DISABLE is set in config it is no longer possible to eanble transmit by pressing long TUNE. Pressing TUNE long now is a temporary setting which can be used additional.

vy 73
Andreas

closed 2016-03-13 by df8oe