df8oe/UHSDR #142: Refactoring: Menu Code

db4ple, 2016-03-07:
I am going to refactor and extend the menu code to make it easier to define/change menu structures.
Planned is first to more or less provide the old menu but with new code base. From this, new menu structures can be build. 
Main part of this refactoring will be to provide the ability to fold/unfold of sub menus (menu groups). 
Menu definition will be more robust (introducing new items and reordering menu items will be very simple, just  define and insert item.

Downside is, I would ask to refrain from larger changes to the menu code for a few days. It is no problem to work inside the existing functions, just don't reorder or move code out. Would help me a lot. Should be over or cancelled in a few days. We'll see.

df8oe, 2016-03-08:
OK - not a problem. BTW I agree to completelyrestructure menu...

No hard splitting user menu / configuration menu.

Menus are splitted to their function:
- display
- audio
- TX (power, SWR, I/Q for TX e.g.)
- RX ((filters, gain, mode-presets, I/Q for RX e.g.)
- misc (voltageadjust, button swaps e.g.)

This will be the first "screen". Change the selection ith encoder #2 - but not cursor goes up and down - it stays at one position.Entries are scrolled up/down,selscted is fixed e.g. at position 3. You select the entry by pressing F4 (which is now function "OK"). So you enter selected menu. Inside this you can select item like above and  adjust with encoder #3.

The depth should only be *2*. No numbers - only description of function.

Additional there is a user defined menu where user can put 20 individual items he selects out of all menus. He can add a entry again with "F4" - which adds an item to the user defined menu.

User defined menu can be reached by entering menu. The first item which is selected is always available and is named "config". With this you enter main menu I described first. The other entries are user-defined. In a later version we can think about user can restructure the items in his personal menu.

vy 73
Andreas

db4ple, 2016-03-08:
There will be a separation between menu structure and menu rendering on screen.
First we change the first part, and keep more or less the old "rendering", then we change the way the user interacts with the menu. OK?

df8oe, 2016-03-08:
OK! I wrote this because of may be these informations are important for the way of rendering and splitting functions. It is not neccessary to do twice same work ;)

vy 73
Andreas

pe1jzq, 2016-03-08:
Hi All,

When doing the menu structure, can I make a sugestion:

Is it possible to use the dail encoder ipv enc dail3 for selecting value 
I noticed it by myself that I use accedenly the dail instead enc3knob.

Just an Idee
regards

pe1jzq

df8oe, 2016-03-08:
I dont think thats a good idea because of many settings want to see that frequency changes can be done simultaneously so this encoder must rest at its normal function...

vy 73
Andreas

db4ple, 2016-03-09:
First preview in pull request (needs to be enabled). Not yet usable at all.

db4ple, 2016-03-10:
OK, now I have something to test, see pull request #146 .
While it is for sure not complete and some features are missing (e.g. auto-advance/back when reaching end of screen) it works and can be used.

db4ple, 2016-03-11:
Code now seems to be pretty well working, now needs some use. See #148 
Menu structure:
Restructured the menu grouping. This is intended to
be just the starting point. 
I would like others to decide about the menu structure (based on discussions)
and then to implement it. This will help to figure out
if it is easy enough to do this without cause trouble.

df8oe, 2016-03-13:
Main work finished. Now fine tuning starts...

vy 73
Andreas

closed 2016-03-13 by df8oe