df8oe/UHSDR #144: Encoders

oz8gw, 2016-03-09:
Sometimes when I turn one of the encoders (for example change the frequency) the radio does not change the frequency on the first step. Sometimes even change one step to the wrong side. This also sometimes happens in the other three encoders. Therefore I guess it is a software-issue.
Vy 73 de OZ8GW, Leif

db4ple, 2016-03-10:
Yes. Most likely an interaction between software and hardware. DL2FW is looking into this I think, as part of his dynamic encoder "code tuning".

db4ple, 2016-03-11:
DL2FW committed his code today, and it seems to work much better than the code before. Did not notice "reverse" steps anymore.

oz8gw, 2016-03-12:
Hi again.
Nice to hear.
Now I look forward to see the next uddate with the improvements.
Vy 73 de OZ8GW, Leif.

closed 2016-03-12 by oz8gw