df8oe/UHSDR #157: Color

HB9LFU, 2016-03-14:
Hello mcHF community,

you have made a great job with the most recent firmware changes: a wide variety of filters, responsive encoders, well structured menu and more. 

Just some trifles:

Could there be change of color for the RIT when it is not zero? 

And could the MODE be displayed in white letters on a blue background for better
readability (especially for the elderly or those with impaired sight)? 

vy73s
Roger

df8oe, 2016-03-14:
done.

vy 73
Andreas

closed 2016-03-14 by df8oe

HB9LFU, 2016-03-14:
Thanks Andreas, I usually do not use devel firmware but it is running so well that it is a joy. 
I even switched DSP in the menu to on and had no trouble so far. 

vy73s
Roger

df8oe, 2016-03-14:
I hope we can push this devel to testing stage this week. Speed of development is really fast so that it is not easy to find a state where you cantell: "That's it for the moment..." :)

DSP is instable. If you use it in normal operation with band switching, doing QSOs, mode switching it often happens that mcHFpowers down without any reason. You canhave luck thatit does not but this is not a stage we can offer this as a stable function for operation. We must have a look what is going on there - time will come to do this ;)

vy 73
Andreas

HB9LFU, 2016-03-14:
Andreas, so far no power down because of dsp! But I always switch it off the moment it is not required anymore. I don't hop around the bands with dsp on.  This may help.

vy73s
Roger