df8oe/UHSDR #173: Encoder without touch

S52CQ, 2016-03-21:
Hi,

I am testing ver. mchf_19_03_2016_153156..bin

With comparison to previous versions I can see heavy problems with encoder reaction time. Let me point out that I did not make Touch screen mod (yet), because I have a nice enclosure with a plexiglass window.

Does the fact that I don't have touch screen mod have anything to do with encoder problems?

73 de Jure, S52CQ

df8oe, 2016-03-22:
Hello Jure,

nothing has changed in code related to this. Touchscreen modification in fact is mandatory to get access to all features - but does not affect encoder speed!

We have cleaned up big parts of code so that chance of unwanted affects if we are working on a special part of code which occur in a quite different part gets more and more smaller. But of course it never will be impossible but now it is acceptable already.

I do not have noticed anything changes in behaviour of encoders in the last time (especially not at the snapshot you mentioned). Encoders are working absolutely fast and much precise than it was before March 2016 :)

vy73
Andreas

closed 2016-03-22 by df8oe

S52CQ, 2016-03-22:
Hi Andreas,

you know that I would not report an issue if it was not there. My encoder is having serious problems with reaction times. I can record a video if you'd like to see that. Older versions did not have that problem, version mchf_07_03_2016_072705..bin is working OK with respect to encoder.

73 de Jure, S52CQ

df8oe, 2016-03-23:
Hello Jure,

work on encoders has not started yet. They are reacting sometimes better, sometimes not so good - result of improving and implementing features to firmware. Because of cleanup of old code is absolutely not finihsed it cannot be guaranted that encoders work every time the same. If reaction is absolutely bad we fix it immediately. At this stage no one of the coders and not one out of our project group notices malfunction so it is not an important change that is recognized by a majority. Because of it is absolutely possible that encoder reaction time changes with every commit (!!!) until code cleanup is finished there will be no bugfix instead if heavy malfunction - which does not exist. I am working with my four mcHFs and I have not noticed changing in encoder behaviour.

Please be patient until code cleanup is finished - which (I hope) will be before summer 2016.

vy 73
Andreas

S52CQ, 2016-03-23:
Hi Andreas,

please don't take that I am unpatient in any way - I am **very ** happy with mcHF, current version (mchf_07_03_2016_072705) works for me great and I thank all of you guys for great work that you've done. 
I've simply reported an issue in ver. mchf_19_03_2016_153156. 

73 de Jure, S52CQ

df8oe, 2016-03-23:
Hello Jure,

I always have seen your report as bug report - nothing else! And it is appreciated to give such reports (now and in future). But no one who does not work on code himself can know what is "normal" and what is a bug which has to be fixed. My explanation only should explain to you that these changes in encoder behaviour is absolutely normal at this stage and can improve or can get worse in a specific bandwidth with every commit. At this stage time between two encoder polls is build up by processing time of filters, audio, LCD and if you change something there encoders change, too. Until this is cleaned up and all functions will be timed by a real timer this will remain...

Your reports are welcome and if here is a bug hich can be removed we will do this ASAP.

vy 73
Andreas