df8oe/UHSDR #177: VFO A and B

oz8gw, 2016-03-24:
The mcHF is tuned to different frequencies on VFO A and VFO B no matter which band.  If you then switch to another band  and then switch to the opposite VFO and then go back to the first band, then VFO A and VFO B is tuned to the same frequency.  It must be a bug.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-03-24:
Hello Leif,

may be one of the VFO storage places on one band is uninitialized. I use split (VFO-A / VFO-B) very often and I cannot see any issue. But if one VFO never was used before on one (or more) band(s) that might be a problem (I cannot reproduce because both VFOs are used on all bands on all my mcHFs).

For testing please step in this way:
1) set VFO-A and VFO-B to different frequencieson each band
2) store this state by press&hold MENU or power down mcHF via power button.

So every storage place gets defined value. Thats the only reason I can think about if you do see an issue there.

Please let me know if this works for you.

vy 73
Andreas

oz8gw, 2016-03-24:
Hi again.
I have testet the way you suggested. The result is exact the same as I wrote from the start.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-03-24:
We must wait if someone else tests this. I cannot reproduce here.

vy73
Andreas

df8oe, 2016-03-24:
OK - I have reproduced.

You changed bands while VFO is set to "B".

I never have had done that before  - always have had switched back to "A" before changing ;)

I will have a look. Many thanks for your report!

vy73
Andreas

df8oe, 2016-03-24:
fixed.

vy 73
Andreas

closed 2016-03-24 by df8oe