df8oe/UHSDR #191: Menu

oz8gw, 2016-03-25:
As far as I can see "Wide Filt in CW Mode" does not work. No matter if it is on or off I can only choose between 300 and 500 Hz in CW-mode.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-03-26:
Hello Leif,

the two menu items
"use wide filters in CW mode"
"CW filters in SSB mode"
are obsolete from now on.

You can select specific filters which are preferred ones for each mode with a short press on G4 as before. In next time there will be added menu where you can select what filters you can toggle in what mode (t this stage it is predefined in firmware).

If you want to leave this mode specific preset you press G4 long time--> filter display changes to orange. Now you can quickly step through ALL possible filters by turning encoder 3 to get access without any regressions. If you want to leave this mode press G4 short again.

vy 73
Andreas

closed 2016-03-26 by df8oe

oz8gw, 2016-03-26:
Hi Andreas.
Thanks for the answer.
I wonder if this new way og choosing filters is in the next FW-release. 
I am using testing version 0.219.27.7.
Can you tell me if there is manuals for versions newer  than 0.0.219.26, 
and where to find it.
Vy 73 de OZ8GW, Leif.

Den 26-03-2016 07:42, DF8OE skrev:

df8oe, 2016-03-26:
Hello Leif,

at this stage there is no documentation available beside information in history.txt located on GitHub. When next master (stable) firmware is released we will put all new and/or changed ui functions in GitHub WIKI. Until then you must dig (or ask) for specific operating steps. In devel ui functions sometimes changes behaviour many times a day and are not "finished" so that there does not exist any manual.

vy 73
Andreas