df8oe/UHSDR #201: White Screen when using firmwares higher than 219.27.4

HB9LFU, 2016-03-28:
Hello mcHF programmers

because your devel firmwares work very nicely I have used them from very early on.

In one mcHF the display is simply just white when I go beyond firmware 219.27.4

When I return to 219.27.4 everything is all rigth again. 

vy73s
Roger

df8oe, 2016-03-28:
We have two or three mcHFs out of > 100 in our german group with the same problem. We are not sure if LCDs are 'out of specs', defective, "clones" or do have other LCD controllers. There exits a different init code which was sent from hotmcu.com (because the old one "has bugs"). But using this init code causes other LCDs, which are wokring fine with existing firmware, to show a white screen.

At this stage we just look what is happening out in the world but cannot do anything. It is a LCD issue.

One hint what you can try:
Problem mostly appears when working in parallel mode. Swap to SPI mode and take a look if it works there - feedback is welcome!

vy 73
Andreas

HB9LFU, 2016-03-28:
Hello Andreas,

here it is one mcHF out of four. Only one shows this strange phenomenon. Yes, I do use parallel mode. Will try this mcHF with a different spare display in the days ahead. If unsuccessful will try SPI mode after that. Thanks!

vy73s
Roger

df8oe, 2016-03-28:
Feedback is welcome! We are watching this strange behaviour with great interest. If it works using SPI please let us know Controller-ID (read out from "Hardware Menu").

vy 73
Andreas

df8oe, 2016-03-28:
And additional:
I am working on init code actually using my raspberry pi and python scripts for testing edges "on the fly" with interpreted code (much faster than compiling, flashing, next one...) So it would be nice to get a feedback if actually daily snapshot works again for you (please copy timestamp of build to your feedback).

vy 73
Andreas

HB9LFU, 2016-03-28:
Andreas, I can try further devel versions as it is easy to go back to the old functioning 
27.4 version.

vy73s
Roger

df8oe, 2016-03-30:
This problem is hopefully fixed now.

vy 73
Andreas

closed 2016-03-30 by df8oe