df8oe/UHSDR #224: Issue: How CAT Changes  TRX Settings May Cause Confusion

db4ple, 2016-04-01:
Right now the changes made through CAT mode directly affect the settings of the band the receiver was in before starting the CAT request. This has good and bad effects. The bad one is that the frequency and mode of this band memory can be set to any permitted frequency (not just in this band) and mode. This will be stored also after turning off. With the "strange" effect of seeing for example a band in the wrong position when using Band Up/Down. E.g. you are in 40m, set via CAT 3.588 Mhz. You will see 80m in the display but value is stored in the 40m band. When using the Band buttons you will see 160m, 80m, 60m, 80m with 3.588 Mhz, 30m, 20m ... 

Finding a solution which suites everyone is not as easy as  it seems. Simply storing the value in the correct band seems the right way to do it, until you use something like the WSPR bandhopping which will then "kill" all your manually selected frequency and mode settings as it walks through the bands. For this use case the current behavior is better  (not good), in this case you may not want anything the CAT interface does being stored permanently. Other use it together with software to support manual operation and want exactly this. 

So we have to decide how to proceed and how to implement a fix.

df8oe, 2016-04-01:
My proposal:

If first CAT setting command is transferred to mcHF a new "working space" containing actual settings of VFOs and mode is opened and will be used from this point when working via CAT. These changes will NOT affect "normal settings" of the band we have been started. You can play around in CAT as much as you want. When powering down mcHF or changing bands with buttons on mcHF these changes are *NOT* stored.

If you want to store the changed settings on actual band you explicitly have to press&hold MENU button.

I think with this everyone can be happy...

vy 73
Andreas

HB9LFU, 2016-04-01:
It is a good idea not to disrupt the stored  frequencies, modes and filters through cat operations.
On the contrary, it would be a great idea to have both a default  cw and default ssb frequency/mode/filter selection for each band stored to be recalled easily by pressing a button. 
However, there may not be too many buttons left for such luxuries.

vy73s
Roger

df8oe, 2016-04-15:
"Sandbox" option for CAT implemented.

vy 73
Andreas

closed 2016-04-15 by df8oe