df8oe/UHSDR #233: Menu: Change storage  Image rejection for mode to Image rejection for frequency

PA3CCE, 2016-04-06:
I like the new IQ balance adjustment,
now I can turn the image completely into the noise !

In the menu, storing of the IQ balance settings is provided
for the different modes LSB, USB etc.

In my case, I find no- or neglectable differences in
settings between the modes, but quite some difference
in the settings over the entire frequency range.
If confirmed, I should like to have the IQ balance in the
menu per mode, changed to settings per frequency
(Low-, medium-, High Frequency, or even better per band).

Jos (PA3CCE)

df8oe, 2016-04-07:
Hello Jos,

this is a very good idea. When I first adjusted IQ with the new functions I had to increase power because I could not measure unwanted signal! And so I stated that there is a dependency to power, too. If you adjust it @0.5W and signal is completely gone, you switch to 5W and adjust that signal again is completely gone, then switch back to 0.5W you will see there is a slight misadjustment.

But we are talking about things which before nobody was able to see :)

We will think about improved structure, freuency dependency included. It is not high priority because now its working for the first time, but "good things can be improved" of course :)

vy 73
Andreas

PA3CCE, 2016-04-08:
Well, I don't find the present situation real "good", when set at mid frequency,
the image rejction at low- and high-frequencies are rather "poor"

73 de
Jos

df8oe, 2016-04-08:
...but *much* better than all the years before :) So I call it 100% "good". Hundreds of users have lived without a working IQ-adjustment and nearly nobody has complained before. But now, if it works - it is *not good* ??

Sorry - I cannot understand.

vy 73
Andreas

PA3CCE, 2016-04-08:
> ...but /much/ better than all ther years before :) So I call it 100%

A change from not working at all to 100% working for just one frequency 
is huge.
Working for the entire frequency range is  much better.
I think that many users just calibrate for one frequency and think " OK 
fine".
They  seem not to be aware what happens at  other frequencies.
Besides, I never heard you say before that the image rejection was not 
working at all,
or did I miss that ?

Best 73 de
Jos

df8oe, 2016-04-08:
Hello  Jos,

haven't you read many posts in the past in Yahoo-NG that IQ-phase adjust does not work?? These posts go back to the time before I joined the project in May 2015... So the best adjustment in the past was not so good as if you adjust it with actual firmware and go to the band with the worst rejection. Because of it was not working properly before many users never have had adjusted their iq settings.

Additional problem may (again) be we reach edge of mcHF. The problem results of unsymmetrical capacities / inductivities at i and q LO signal line. I have already tested and they depend on many things like voltage of power supply, actual output etc. so it is possible that it is impossible to get all these dependencies included.

vy 73
Andreas

PA3CCE, 2016-04-08:
> Hello Jos,
>
> haven't you read many posts in the past in Yahoo-NG that IQ-phase
> adjust does not work?? These posts go back to the time before I joined
> the project in May 2015... So the best [...]
I agree, I also complained in the past.
  With me I could adjust the amplitude, but the phase adjustment did not 
react at all.
  However, this is'nt "not working at all".
To my opinion this is not to be used as a reference to declare the 
present situation as "good".
> Additional problem may (again) be we reach edge of mcHF. The problem
> results of unsymmetrical capacities / inductivities at i and q LO
> signal line. I have already tested and they depend on many [...]
Sure it will never be perfect, but what I proposed might be an 
acceptable  approach.
Do not underestimate the importance of the best image rejection in Tx- 
as well in Rx-mode.

73 de
Jos

vy 73

DD4WH, 2016-04-08:
Hi Jos, hi Andreas,
my impression is, we have all fully acknowledged that we need a smart solution for the dependencies of the phase adjustment in order to get the very best out of our hardware. But we should also be smart in using a slim solution, if it is feasible and appropriate. I am not sure whether a phase adjustment for every band would achieve this (and would be necessary), but we would probably need more data on that in order to be able to decide.

Could you two state the amount of sideband suppression that is lost when you adjust phase & amplitude perfectly for one band and then change to the band with the worst suppression? Is it in the range of 2-5dB or in the range of 10-20dB? And please also state whether your measurement is for TX or RX. 

Theoretically, (for TX) if we can adjust sideband suppression to an optimum of 55dB for one band and this gets worse for another band in an order of 6dB, well, then don´t let us talk about that again . . ., that´s still 49dB of suppression (10W -49dB equals 126µW, which is acceptable in my view).

For RX, with listening to DX signals and a local 100W station at 12kHz of your frequency (if you use the 12k IF), it could be nice to have 6dB more suppresion . . . however, the best hardware phasing receivers hardly ever achieve more than 45dB of sideband suppression (in the best band!).

Let us look into the HARDWARE signal I&Q chain, this is different for TX and RX:
RX: 
C68/C69 (22nF)--> they should be matched! I did that when I built the mcHF and I do not have such a big improvement with the new phase adjustment as others, maybe because of the matching, dont know!?  
C71/C73 --> must be 0R !
C26/C31 (470nF) --> maybe matching helps? Have not tried yet
C18/C19 (1µF) --> maybe matching helps? Have not tried yet
--> For RX, it should be easy to match the three pairs and have very good sideband suppression with the hardware alone

TX:
C7a/C8a (470nF)
C7/C8 (470nF)
C109-C112 (10µF)
C114-C120 (100µF)
C84-C87 (22nF) --> at least these must be matched!
--> There are not so many capacitors and no inductor in the I&Q signal chain. So, I do not see that it would not be possible to improve a lot on the hardware side by matching the capacitors in the I&Q TX line. 

Question 1: Maybe this also helps to decrease the frequency dependence of the phase adjustment?

Question 2: Is it necessary to distinguish between phase adjustment for LSB & USB, AM & FM?
--> At least theoretically, I do not see any reason for this: All the demodulation steps occur in software. So, if we adjust I& Q phase and amplitude before demodulation (as is the case in the moment), this adjustment should be the same and should hold for ALL demodulation modes.
(It would be good, however, if that could be supported by your observations!) 

This could free up EEPROM and menu space for frequency dependent adjustments!? --> Just a thought . . .

73 de Frank DD4WH

PA3CCE, 2016-04-08:
Hi Frank,

First of all I must tell you that at this moment I cannot perform any 
test, since
today my mchf stopt working. Something is wrong in my LO.
At times the frequency display is in red color and the spetrum display is
interrupted in abt 1 sec rate.
When I change frequency, the spectrum display stops inmediately.
Could not find the cause so far.

I also can tell you that I have made all modifications as they are in 
the circuits
of IZ6MAF, so all unnecessary capacitors are shorted etc.
I did the tests in Rx mode only.
Further more I had matched the appropriate components in the demod circuit.

I can give you the adjustment figures  below :

Freq                   LSB                            USB
                     ampl/phase            ampl/phase

  3MHz                 5/2                         5/1
  5MHz               11/5                      11/4
  7MHz               17/8                      18/7
10MHz           19/14                    20/13
14MHz           30/22                    28/22
21MHz           50/38                    54/37
24MHz            58/52                   58/50
28MHz            63/65                   63/65

These are the settings I found in the menu (all positive).
You see that the difference between the demod modes are neglectable,
this was the reason I did not test the other modes.
You also can see large differences per frequency, in phase as well in 
amplitude.
I did not write down the differences in image rejection but, since you 
know the
change per step, you can have an idea.

As soon as I have mcHF working again I can profide you more info.
I hope this helps.

best 73 de
Jos

DD4WH, 2016-04-08:
Hi Jos,

thanks a lot for your data. That helps, also it would be very good to also have settings translated into dB of suppression ;-) I altered the setting of the finetuning a few versions of firmware ago, so a difference of 5 to 63 is not so big anymore as it was in former times ;-)

I had the same problem with red digits and it returns from time to time. Check soldering of the Si570 and around that, very likely that the problem is there, but Andreas probably knows more.
From your data, I would conclude, we can stop distinguishing between modes in the phase and amplitude adjustment! Also, if you have matched capacitors in the Rx chain, it would have to be concluded that this does not help against the frequency dependance of the adjustments! Very interesting and valuable results. I will think about that and wait for more info from other users.

Thanks a lot for your info and good luck in fixing your rig! If it is ok, I would appreciate very much more info, eg. from TX mode and other measurements!

73 de Frank

df8oe, 2016-04-08:
Hello Jos,

that is very interesting! My settings differ only +/- 2 (both phase and gain) between optimum and worst band.

Why does yours differ so much? Now I understnad your opinion - but I think, you do have a hardware problem. All of my 4 mcHFs differ only +/- 2...

y 73
Andreas

PA3CCE, 2016-04-09:
Hello Andreas

+/- 2 difference over the entire frequency range, are you sure ?
That is almost ideal, I can hardly believe that.
I hope my mcHF will be operable soon, I ordered a new RF board,
the old one is, due to many experiments, worn out.

best 73 de
Jos

df8oe, 2016-04-09:
Yes. I have one mcHF built as RX-only and three mcHFs built as transceivers.

but I have made a modification on all of these which is not documented yet - maybe that is the reason??!!

I kicked off drivers U12 and U13 on rf-board and swapped R22...R25 with 0R. Drivers are producing more phase shifting at higher frequencies and in my eyes are not neccessary (and maybe they do disturb more than they are useful). But I have not made intensive measuring. Of course this can be the reason for the great differences between your mcHF and mine!

Must be investigated... But I am only one person and have to make business, too , so I do not have much time for all of this. McHF takes already big part of my time :)

vy 73
Andreas

DD4WH, 2016-04-09:
Very impressive differences! Hmmm . . .

Andreas, is that in RX or TX or both? Are the differences between modes also so small?

Would love to try your mod, did you solder out the SN74LVC2G17s or did you just solder two wires on top of them to bridge the buffers?

73 de Frank

P.S.: Unfortunately, I am not able to measure myself, because I have no stable RF generator with such a high drive . . .

df8oe, 2016-04-09:
You must desolder little drivers and connect in/out (2x at each driver) with wire. My settings are in a small range at RX and TX.

vy 73
Andreas

PA3CCE, 2016-04-09:
@ Andreas, Frank,

OK Andreas that is most interesting information, thank you.
As a matter of fact, the new RF board I ordered is of version 0.5.
In this version these drivers are rejected indeed and the traces
of the LO are matched for impedance, as described by Chris.
As soon I have the results, I will let you both know.

Have a nice weekend,
Jos

df8oe, 2016-04-09:
Hello Jos,

thank you for this information. That will be an explanation... Additional I have used 7474 manufactured by "Potatoe Chips" which has anupper working frequency of 1GHz. This will become eccessary when you will add the rf-mod-pcb so that 6m and 4m are available, too. Maybe timing at higher frequencies is more precise like on normal logic IC...

Your feedback after getting new pcbs is welcome!

Have a nice weekend, too
Andreas

PA3CCE, 2016-04-09:
Hi Andreas

Thanks again, I must have some "Potatoe Chips" circuits somewhere too, 
so I will see.
I hope and expect theat I can keep using the UI version 0.4, for I just 
ordered the RF board.

Cheers
Jos

DD4WH, 2016-04-10:
Very interesting, these 7474 chips! Did you also use the faster potatoe version of the 3253 as substitutes for the 3253 Muxer? The potatoe version seems to have much larger on resistance (17-22ohms vs 3-6ohms), could potentially be a problem! 

73 de Frank

DD4WH, 2016-04-10:
I could now reactivate my Dipmeter and an RF amp and an attenuator, so I had an RF source at hand which produced  a huge RF signal with -17dBm. Here are my settings for optimum IQ amplitude and phase adjustment setting for specific frequencies:
(frequency, (ampl., phase for LSB), (ampl., phase for USB))

2.5MHz  -14/-20 LSB  -14/-20 USB
4MHz  -13/-19 LSB  -12/-19 USB
8MHz  -13/-20 LSB  -13/-20 USB
12MHz  -20/-34 LSB  -23/-23 USB
16MHz  -16/-30 LSB  -16/-27 USB
21MHz  -24/-31 LSB  -25/-29 USB
24MHz  -34/-49 LSB  -35/-44 USB
28MHz  -37/-55 LSB  -38/-46 USB

side observation: at 24MHz and 28MHz the nulling was dependent on where I put my fingers on the mcHF (which has no shielding or metal case in my mcHF)!!! That could also be one of the reasons for your varying results, Jos!?
 
My conclusions:
1.) we do not need to distinguish between LSB, USB, AM, SAM, FM in the amplitude & phase adjustments --> software can be simplified
2.) the adjustment is more or less frequency dependent (depends on your build/your measurement situation) --> whether we need a band-specific possibility to adjust, is not clear yet, because in my case (see above), it would really be overkill . . . but in others (Jos!) it could be nice to have.
3.) The frequency dependance is probably 100% the result of the buffer ICs U9, U12, and U13 (if you compare Andreas´ results with Jos´ and my results). The buffer ICs U9, U12, U13 and corresponding Rs R19, R22-R25 should be eliminated as a suggested modification.

Comments and more measurement data very welcome!

73 de Frank

df8oe, 2016-04-10:
I do not have swapped mixers (due to quite different Ron what will be a problem). But I have tested nd they are usable up to 100MHz - they will work with 6m and 4m od, too.

vy73
Andreas

PA3CCE, 2016-04-10:
Thanks, see below.
Jos

> I could now reactivate my Dipmeter and an RF amp and an attenuator, so
> I had an RF source at hand which produced a huge RF signal with
> -17dBm. Here are my settings for optimum IQ amplitude and [...]
Jos :
I had a shield between RF board and UI board, but no metal case.
Observed NO sensitivity to fingers placed on the mcHF, measurements were 
stable did not look for instabilities.
My measurements  were executed at approx. -40 dBm at antenna in.
Had a RF preamp with PGA103+.
>
> My conclusions:
> 1.) we do not need to distinguish between LSB, USB, AM, SAM, FM in the 
> amplitude & phase adjustments --> software can be simplified
Jos:
Agreed
> 2.) the adjustment is more or less frequency dependent (depends on
> your build/your measurement situation) --> whether we need a
> band-specific possibility to adjust, is not clear yet, because in [...]
Jos:
Overkill ? , I don't know,  what was your difference in image rejection 
between lowest/highest band when adjusted at mid band?
Sorry I did not note down ,  as said I could adjust deep into the noise, 
so rejection was >>-60 dB.
I prefer to have highest rejection as possible to avoid all kind of 
interference.
>
> 3.) The frequency dependance is probably 100% the result of the buffer 
> ICs U9, U12, and U13 (if you compare Andreas´ results with Jos´ and my 
> results). The buffer ICs U9, U12, U13 and corresponding Rs R19, 
> R22-R25 should be eliminated as a suggested modification.
Jos:
U12 and U13 were still in, corresponding Rs were out.
I'm anxious to see the results of version 0.5 RF board.
However, results may be dependent of many variables.
>
> Comments and more measurement data very welcome!
Jos:
Will do so when new version 0.5 RF board has arrived and built.
When the LO modification solves this problem (more or less),
there still must be a solution for version 0.4 users.

DD4WH, 2016-04-10:
I just soldered out U12 and U13 and put in 0R for R22-R25. So these buffers are now out of the I&Q path:
Did the same measurements again for RX:

2.5MHz  -10/-22 LSB     -7/-22 USB
4MHz  -11/-24 LSB     -4/-23 USB
8MHz -3/-29 LSB     2/-28 USB
12MHz 2/-33 LSB     5/-33 USB
16MHz 11/-31 LSB     14/-31 USB
21MHz 4/-44 LSB     7/-43 USB
24MHz 5/-72 LSB     7/-72 USB
28MHz 33/-56 LSB     35/-56 USB

So, what would you conclude from this?
Comments welcome! ;-)
73 de Frank

DD4WH, 2016-04-10:
I would conclude that my measurements at 24/28MHz probably are not reliable, because the (very strong) signal of -17dBm seems to enter the mcHF over additional paths . . .
However the other measurements seem to indicate the following:

@2.) we need a frequency dependent I&Q amplitude and phase adjustment. I fully agree with you, Jos, now with those measurements. A difference in adjustment of +-3 can result in a difference in mirror rejection of >20dB (not measured, but a sophisticated guess by me). Sometimes +2 steps turns absolute silence to a rather loud annoying heterodyne.

@3.) my hypothesis must be rejected: the buffers are not the problem! New hypothesis: the 74LCX74M (U11) is the problem, that chip seems to have a huge frequency dependence in handling amplitudes and phase. I will order the "wonder chip" (7474) by Potatoe chips in order to validate that.
Andreas, could you confirm, it is this one, please:
http://www.ebay.com/itm/7474-G-Series-GHz-TTL-CMOS-logic-IC-14pin-SOIC-QTY-1-/330551715157

73 de Frank

df8oe, 2016-04-10:
YES - that is the one I am using.

vy 73
Andreas

PA3CCE, 2016-04-10:
>
> I would conclude that my measurements at 24/28MHz probably are not 
> reliable, because the (very strong) signal of -17dBm seems to enter 
> the mcHF over additional paths . . .
Might be, but see my findings also.
Can't you make a simple attenuator of say 20dB or more ?
OK, you are using a dipmeter, may radiate a lot :(
1 turn loop on a long cable ?

However the other measurements seem to indicate the following:
> @2 <https://github.com/2>.) we need a frequency dependent I&Q
> amplitude and phase adjustment. I fully agree with you, Jos, now with
> those measurements. A difference in adjustment of +-3 can result [...]
This matches my findings, 1 step makes a difference of no tone to a 
noticeable tone.
I also could make use of the S-meter, no signal was approx. S3 of noise.
Signal input was S9 + 30 dB or so.
I could calculate the approx. rejection  1 S-unit = 6dB
>
> @3 <https://github.com/3>.) my hypothesis must be rejected: the 
> buffers are not the problem!
Looks worse than better, but as said there are probable more variables, 
it's tricky.

New hypothesis: the 74LCX74M (U11) is the problem, that chip seems to 
have a huge frequency dependence in handling amplitudes and phase. I 
will order the "wonder chip" (7474) by Potatoe chips in order to 
validate that.

DD4WH, 2016-04-10:
After some time for thinking, I have some new findings:

1.) With the buffers in my mcHF, there were no differences between LSB/USB. After taking out the buffer ICs, measuring, and after that additionally resoldering my solder bridges and measuring everything again, there are still significant differences between USB & LSB, but only in the amplitude adjustments, not in the phase adjustment! Thus, the buffers U12/U13 seem to be useful for eliminating frequency effects on the IQ amplitude differences. So, we do need software diiferentiation between LSB/USB adjustment, because RF board 0.5 has no more buffers.

2.) & 3.) Even IF the potatoe chip 7474 eliminates frequency dependancy, probably not every user will order it and build it in. So do we want a software adjustment? If yes, I would suggest: 12 different adjustments with the following low corner frequencies which I have chosen for ham radio use AND SWL in mind:
100kHz - 2200m
450kHz - 630m
1.75MHz - 160m
3MHz - 80m
5.5MHz - 40m
8MHz - 30m
11MHz - 20m
14.5MHz - BC 19m
16MHz - 17m
19MHz - 15m
22MHz - 12m
25MHz - 10m 

12 adjustments for amplitude & phase for USB/LSB separately means 48 adjustments (and also 48 EEPROM memory places)!

73 de Frank

df8oe, 2016-04-10:
After we got a working IQ-adjustment -and first at this point - we consider that there is huge hardware problem sleeping in design of PCB. Because of IQ-lines are not designed @50 ohms there are frequency dependencies which never were recognized before. Capacities of mixer inputs and output impedance of drivers (or of 7474) are involved, too.

I think before we start a huge storing orgy we first have to accumulate informations. Maybe it is possible to find a formula how better scalings can be calculated. But therefore we need much more samples of actually best IQ adjustments...

vy 73
Andreas

DD4WH, 2016-04-10:
Yes, you are right, the new software now makes the hardest hardware tests possible for the first time!

And the few results are still not leading into one direction and hardware effects are complex. 

So let us do more tests and gather more data under the diverse configurations.

I am really curious whether the 7474 chip from Potatoe chips will solve all that . . .

73 de Frank

PA3CCE, 2016-04-10:
@Andreas, Frank

Just my penny :

1)   I do NOT see the necessity of making difference in adjustment in  
demodulation modes.
        dependency of other factors are many  many times  bigger.
        Besides, I have never seen this in other projects either.

2) I do NOT believe in one  satisfactory adjustment for all frequencies 
(and on all existing mcHF variants) ,
     there are too many dependencies  involved ( trace impedances, stray 
capacitances, stray inductances,
     component variations, etc. etc.) .

Best compromise :
Be realistic and make the best (simplest) solution for everyone, provide 
for an adjustment for each
frequency band (maybe more for the highest frequencies), NOT regarding 
modulation mode.

Best 73
Jos

df8oe, 2016-04-10:
It is not trivial to implement so much storing (especially for users who don't have fitted serial EEPROM because in actual virtual EEPROM there is a cut at 383 words).

So let us "be patient" and find a solution.

Again:
all these problems came up a few days ago when IQ adjustment works as fine as is is now....

I don't think it is a problem in waiting some more days (or weeks) for finding best compromise - problem exists from the birth of mcHF on!!!!

We are pointed on this and working on it to get best solution.

vy 73
Andreas

PA3CCE, 2016-04-10:
It's your choise.
How many stores do you have now, for all modulation modes !
When you find a better solution you can change any time.

Jos

df8oe, 2016-04-10:
We don't have enough at the moment for users without serial EEPROM.

vy 73
Andreas

PA3CCE, 2016-04-10:
how do they cope now with all these modulation mode stores ?
No argument, sorry.

DD4WH, 2016-04-10:
Keep calm, guys ;-) Rome has not been built in a single day . . .

We have all understood the problem and the arguments now and now it is time to stop talking, to measure again, to think hard again for the best optimization (because we are talking about an already superb mirror rejection here --> in the early days of the SDR cube 25dB was called "very usable" and we have >60dB when propery adjusted!!!), and then to start programming with an economic view on the EEPROM storage room :-)

Have fun!

73 de Frank

PA3CCE, 2016-04-10:
OK Frank,

No problems here, I just like real arguments ;)
Will wait and see patiently what happens.
Will let you guys know when I have more info.

Jos

DG9BFC, 2016-04-13:
what about storing some set points and the mcu calculates the points in between??

say you have a point of phase /gain at 1 megs of 20/30 and another at 6 megs with 25/35 ...

and you find out that the gain and phase is almost linear over that range (means at 2 megs you need 21/31 and at 3 megs 22/32) ... then you just store a few rough set points ... and rest is done in a calculation step inside the mcu

if at lower or higher bands the phase and gain correction is not that linear ... then just spread in another set point .... guessing that with a few points an almost ideal curve can be stored ... without the need of a too high number of set points

df8oe, 2016-04-14:
That's the idea we have had, too! But there exist many possible modifications (e.g. swap 7474 with "potatoe chip", leave uffers like in rev. 0.5 is, different mixers...) All of these will change frequency  dependency of IQ lines. So we will start investigations next weeks and hope as many users as possible will participate. We will offer a spreadsheet where every users fills in his personal IQ settings for many frequencies and modes, added his hardware he uses. Then we hope we can see if a function exist and how to detect if there are different possibilities.

y 73
Andreas

HB9LFU, 2016-04-14:
Hello Frank / Andreas and others,

thanks a lot Frank for sending me the link to the message about signal quality and using the tune button. With a real signal the spectrum is compeletely different and does not show these spurs.

However, there was one left close the desired signal. But I could reduce it drastically with the IQ menu settings. On 80m -50 was ideal. On 10m -50 was fully indadequate, I had to crank the value up to almost +99. The value for 60m was also ok on 80 and 40m. Also 30 and 20m were fairly close, the same holds true for 17 and 15, as well as 12 and 10m. 

vy73s
Roger

DD4WH, 2016-04-14:
Hello Roger,

nice to hear that the problem is solved now! The TUNE function is very
nice, but it is not good for checking signal quality ;-)

And also the phase adjustment does solve many "problems". However, we will
collect more data on that and Andreas is preparing a spreadsheet for
structured data collection and it would be very nice if you (and many
others!) could participate. It will be prepared in the next weeks and
announced in the Sulingen forum.

Have fun with the mcHF!

73 de Frank DD4WH

HB9LFU, 2016-04-19:
Hello Andreas,

this eeprom is so cheap and it does not take an hour to open the case and solder an eeprom to the ui-board that this part should become a mandatory thing. 

vy73s
Roger

PA3CCE, 2016-05-22:
Hello Andreas
Please find below the I/Q balance settings I found for my new RF board version 0.5 :

Rx I/Q adjustment : (mirror inaudible)
============================
Conditions :
Vi = -50dBm @ 50E
Rx Preamp = 5dB pad + BGA103+ 
LO divide by 4 (U11) = Potatochips PO74G74A
mcHF UI board version 0.4
mcHF RF board version 0.5
All IZ6MAF modifications implemented.
Firmware 1.1.0 / BL 1.0 (May 20.. 21:06:47 EST)

3.65MHz 
Amplitude :. ...USB..10.....LSB  10
Phase : ..........USB  -15....LSB -14
	
 5.250MHz
Amplitude :.....USB..20.....LSB...18
Phase :..........USB.-14.....LSB .-10

 7.10MHz
Amplitude :......USB..22.....LSB..18
Phase : ...........USB.-11.....LSB...-8

10.10MHz
Amplitude :.......USB..22....LSB..22
Phase :.............USB..-5.....LSB..-5

14.1 MHz
Amplitude :......USB..35.....LSB.. 32
Phase :............USB...1......LSB....1

18.69MHz
Amplitude :.....USB..42.....LSB..38
Phase :...........USB..15.....LSB..17

21.2 MHz
Amplitude :.....USB..54.....LSB..53 
Phase : ..........USB..16.....LSB..17

24.89 MHz
Amplitude :.....USB..63.....LSB..61 
Phase : ..........USB..22.....LSB..22

28.500 MHz
Amplitude : .....USB..69.....LSB..69
Phase ;............USB..52.....LSB..52

Best 73
Jos

db4ple, 2016-05-22:
Hi,

your measurements are great. It seems to me that there is almost no difference between USB and LSB settings. For those where there is a difference, can you live with the  average between both settings? Would allow us to use just one set of settings. 

What I can see is there might be some formula to calculate the variation across the frequency from the low frequency  and high frequency value. Catch is, we would need more data to do validate that.  

That would limit the number of calibration steps in order to get to a better working mcHF. And we could reuse the existing EEPROM Layout.

73
 Danilo

PA3CCE, 2016-05-22:
Hi Danilo,
Sure I can live with the average between the settings for USB and LSB, these diferences
are marginal.
It would be nice if you are able to implement a formula for the variations by frequency, it would be a big improvement in mcHF's performance.
Of course we need more information from other users to verify.

vy 73
Jos

HB9LFU, 2016-05-23:
Hello Danilo,

is there not enough storage space to have a per band adjustment of these settings?

vy73s
Roger

df8oe, 2016-05-23:
There is. But we want to keep adjustment menu and procedure as easy as possible and would prefer not to dig through a huge number of settings.

vy 73
Andreas

HB9LFU, 2016-05-23:
Hello Andreas,

then why not adjust 80-60-40, 30-20,17-15 and 12-10 together. This would simplify the process considerably. I do not believe it is a good idea to calculate some sort of curve for the whole range and for all the different sets of mcHFs with varying modification levels. 

vy73s
Roger

PA3CCE, 2016-05-23:
Hi all

When a formula alone does not work, a combination of both band grouping and a formula might come close to the ideal.

73 de
Jos

df8oe, 2016-05-23:
We are thinking about all this possibilities. There are BC bands, too and we must find the best solution for all users. We must collect more data (TX, too) for 0.4 and 0.5 PCB and different modification stages so we can find final solution.

vy 73
Andreas

PA3CCE, 2016-05-23:
I/Q adjustment for the HAM bands is a way of band grouping for the BC bands !

73 de
Jos

db4ple, 2016-05-23:
Maybe to clarify: Even a formula approach will need some device specific calibration values, in order to to calculate the intermediate values for  the various frequencies.

PA3CCE, 2016-05-25:
Hi Danilo,  Andreas and others,

Today I compared I/Q balance results from my old version .04 RF board to the last results from the new .05 board.

Amplitude :  5 to 63  (old board)  vs  10 to 69 (new board).
This is a variation of +58 (old) vs +59 (new).

Phase     :    2 to 65 (old board) vs  -14 to 52 (new Board) .
This is a variation of +63 (old) vs +66 (new).

I find this similiarity remarkable.
Take into account that the LO has changed importantly :

1) U11 74LCX74M (old), PO74G74A (new)
2) No buffers 74LVC1G17 and 74LVC2G17 (new)
3) Different LO layout (traces)
4) I used an other specimen of Si570 in version 0.5

So, it looks like the variation over the frequency range is more consistent then I expected.
This gives me confidence in you finding a working solution for all users.

I will give you my results for the Tx part, as soon as available.

Best 73 de
Jos

HB9LFU, 2016-06-14:
Hello Danilo,
as promised here my IQ correction data for 80 to 10m. As I used cw with a key I could only vary USB TX IQ Balance:
80m= 21
60m= 42
40m= 27
30m= -52
20m= -61
17m=-67
15m= -75
12m= -80
10m= -99

regards
Roger

df8oe, 2016-06-14:
...and here my data for the same working condition:
USB TX IQ Balance:
160m: -38
80m:  -37
60m: -37
40m: -38
30m: -34
20m: -26
17m: -12
15m: -10
12m: -6
10m: 1
6m: 20 (measured directly after mixer because filters are not ready yet)

As you see there is absolutely nothing similar...

vy 73
Andreas

HB9LFU, 2016-06-14:
Yes, quite different Andreas!

vy73s
Roger

df8oe, 2016-06-14:
I have four mcHF with a working TX. All four have nearly the same modifications. But all of them have quite different settings. Not only the values: one is going + and  - swapped. All have different slopes, two have a "glitch" wher e setting does do a jump (not at the same frequency).

*Very difficult* to handle...

vy73
Andreas

db4ple, 2016-06-14:
Let me put it in other words: The current approach is not worth being kept. Given the amount of change of the frequency range shown in all measurements a single value makes absolutely no sense.
I am wondering if we can confirm that settings for RX AM, USB, LSB for the same frequency are more or less equal. That would lessen the amount of values being required.
What worries me are the glitches. That kind of thing breaks almost all interpolation strategies since a glitch of that amount will make very strange values all the way to the next known value.
However, any approach which has some sort of per frequency values is better than the current one.

For the storage of the configuration values (as Andreas said) we have to extend our approach to have "EEPROM Only Values". I.e. values which are more or less directly written to EEPROM and always read from there (which is fine for low frequency write value, also  the PA adjust/coupling values fall into that category if you thing of it. We should be able to do that, its not too difficult to do.

73
Danilo

HB9LFU, 2016-06-15:
Hi Danilo,

I do only cw so I cannot confirm that the other modes have the same values on the same band. But from what I heard from others does it not make a huge difference. So it could be ignored.

What I find important is to not  base future developments on those who have not yet installed the external eeprom. This part is cheap and does not take half an hour to install. 

Yes, with the glitches a per band solution seems inevitable. 

vy73s
Roger

df8oe, 2016-06-15:
Hi Roger,

virtual EEPROM is an important fallback for users of serial EEPROM, so it MUST work. The option to backup and restore a set of settings is an important goal of the "EEPROM modification" and a *MUST*. But if we can expand virtual  EEPROM for backup purposes it can be used for those who do not have serial EEPROM without any further changes.

vy 73
Andreas

HB9LFU, 2016-06-15:
Hello Andreas,

you have pushed through the touch screen use, now I think it is time that you push through the use of the external eeprom.

vy73s
Roger

df8oe, 2016-06-15:
Hi Roger,

programming work for creating storage of many IQ datasets is identical if use of serial EEPROM is mandatory or not...

We do not want to loose capabilityof backing up serial EEPROM to virtual EEPROM so we must expand it, too, regardless if serial EEPROM is available or not.

vy 73
Andreas

[48 more replies]

closed 2016-12-12 by PA3CCE