df8oe/UHSDR #243: CW/PTT

S52CQ, 2016-04-11:
Hi all,

I am wondering if CW/PTT CAT control was implemented in any of the dev. versions. Last week I did install one dev. version but CW/PTT CAT control was not working within HRD SW,

73 de Jure, S52CQ

df8oe, 2016-04-11:
All possibilities of FT-817 are implemented - means: whatyoucan do with a FT-817 you can do with mcHF.

CAT-PTT is working - I use it intensively with fldigi, qsstv and wsjtx.

For using CW you need working CAT PTT and possibly USB audio (if CW is generated as "tone"). Both is working.

vy 73
Andreas

closed 2016-04-11 by df8oe

S52CQ, 2016-04-11:
Hi Andreas,

thanks for quick reply. I agree with you that CAT is working - I also use BPSK within Ham radio deluxe (Digital Master 780 within DRM), and I use WSJT-X, that both works great, But CW does not work neither in HRD, neither in TR4W log program or Winlog  that I have tried. I did set mcHF to USB within HRD - and it does turn on TX while no CW keying happens. I did set up mcHF USB interface in the HRD Options. Microphone input is DIGITAL.
If I set ncHF to CW nothing happens - maybe audio does go off for a second but it does not go to TX.

What am I missing that CW works with you and not with me? Did you try with it any LOG program, like TR4W?

73 de Jure, S52CQ

df8oe, 2016-04-11:
You must set mode on mcHF to USB - not to CW... In CW no audio is supported so you only get a "click" at TX start. CW is transmitted in most software via audio I am working CW with fldigi.

vy 73
Andreas

S52CQ, 2016-04-11:
> Za: df8oe/mchf-github <mchf-github@noreply.github.com>
> Kp: S52CQ <stik@svet-el.si>

Hi Andreas, I did have USB on mcHF... 73 de Jure, S52CQ

Poslano iz naprave Samsung

-------- Izvirno sporočilo --------

df8oe, 2016-04-11:
I am running Linux on my computers so I do not know the applications you are talking about (Windows applications...). But it works fine using fldigi - that's the application I mostly use.

Sorry - I cannot help with other applications than fldigi, qsstv, freedv or wsjtx...

vy 73
Andreas

S52CQ, 2016-04-11:
OK Andreas, as I am using Windows LOG SW I will make additional HW to connect RTS/DTR that I get from USB to mcHF and connect CW & PTT to suitable connectors on mcHF. When done I will post to the group solution for all mcHF users that have Windows SW.

73 de Jure, S52CQ

df8oe, 2016-04-12:
Hello Jure,

I cannot understand the problem....

If you want to TX in CW via software:
- programm sends "PTT ON" via CAT
- then program sends CW as audio signal (at mcHF via USB - therefore you need to set mode to "USB" and select "DIG" as audio input on mcHF)
- when TX is finished program releases "PTT" again via CAT

Absolutely simple - you don't need any "DTR/RTS" lines. These lines are not present at mcHF because there is no FTDI chip inside mcHF so usage of these signals will be without any effect. These lines are often used in the time BEFORE CAT was born. DTR/RTS "simulate" a simple switch.

Maybe the programs you want to use are incompatible with the "normal way of transmitting CW via CAT" or you do not have found how to change the settings to this normal behaviour...

There is a huge amount of applications which only works for "special situations" or they are written long ago when PTT via CAT and CW viaaudio does not exist yet. In this case best is to change application (because there are MANY working outside in the world) and not Hardware :) Iknow one rediculous application which persists of using VOX- function and does not use CAT PTT. Programs with such a behaviour are outdated. If the programmers cannot fix this it is TRASH...

Of course if you absolutely will not change to running program and you find a workaround for the application you are welcome to write a howto for our WIKI - maybe thats helps others who want to use same application...

vy 73
Andreas

S52CQ, 2016-04-12:
Hi Andreas,

I did try your suggestion. With all Win LOG programs that I tested it's 
so: I set mcHF to CW (which via CAT is then set to CW within SW) and SW 
LOG program is keying mcHF.

If I set mcHF to USB then LOG program assumes that I am working SSB and 
hence no keying from LOG program. Therefore I have to set mcHF to CW.

I send you screen of the setting within TR4W LOG SW:

So I need RTS&DTR, not only in this SW. But will search also for other 
SW that works. Maybe anyone in German group uses Windows LOG SW that 
works with mcHF?

73 de Jure, S52CQ

df8oe, 2016-04-12:
Hello Jure,

applications that use DRT /RTS are absolutely outdated. Of course you can construct a workaround:

- use FTDI adapter on PC
- connect line you need to PTT
- select "straight Key" on mcHF

With this way FTDI simulates pressing a key... But that is technique of end of the twentieth century :)

give a try to [fldigi...] (http://www.w1hkj.com/FldigiHelp-3.21/html/cw_configuration_page.html)

vy 73
Andreas