df8oe/UHSDR #244: Problem lanching last daily snapshot

wemaus, 2016-04-11:
When lanching last daily snapshot (11.04.2016 13:06:12) I get a problem. After powering up the mcHF the splash screen apears followed by the button test screen with the only to end the test programm and so on...

df8oe, 2016-04-11:
I think you do have a sporadic hardware issue for nothing related to this was changed in firmware.

What button is shown as "pressed" in button test?

I think it is the touchscreen (which is interpreted as "button").

Best choice:
- do the modification to activate touchscreen.

Workaround:
- investigate what pulls down PA4 of MCU (header pin7). Can be flux,a connection to PA4 or so. Pin hasinternal pullup in MCU so that it only can be activvated from outside.

vy 73
Andreas

wemaus, 2016-04-11:
The last version with I have had was the testing version 07.04.16 (943970c). I have downloaded the latest daily snapshot twice with the same result as described above. After the splash screen is visible the button test screen appears for a very short time and disappears without any touch of me. My mcHF had worked very good before and when installing the firmware version from before it also worked very fine again.

A workaround in this case is not possible and I have done all the modifications  which are shown on your site "Modifikationen..." of german forum. 

vy 73, Werner, (noch) DO8WMA

df8oe, 2016-04-11:
Hallo Werner,

bitte neueste Snapshot testen - Problem ist beseitigt.

vy 73
Andreas

wemaus, 2016-04-11:
Hallo Andreas,

nun läuft es wieder wie gewohnt. Danke für die schnelle Lösung des Problems.

vy 73, Werner (noch) DO8WMA

closed 2016-04-12 by df8oe