df8oe/UHSDR #247: Touchscreen

PA3CCE, 2016-04-13:
From start, my dynamic tuning does not work.
Discussed this earlier in mcHF YG.

Finally I received the newly ordered LCD's :
Supplier : Thaoyu Electronics, China.
2 pieces Model HY28B, one delivery.
Setting : 16 Bit parallel.
Firmware : 13/04/2016 09.54.43

White screen : NO
===========

Touchscreen :
===========

Stylus tapped  on Stepsize indicator (1kHz) :
------------------------------------------------

Reaction : NO graying (No dynamic tunining)
Position (in buttontest) 32/46 (Both new display's the same)

Stylus tapped on Watarfall/scope gray bar :
-----------------------------------------------

Left sde : Magnify 1x/2x  (Not very consistent, but it works)
Right side : Scope/Waterfall (somewhat more consistent)

Hardware display :
-------------------------
In hardware info it shows "Touchscreen n/a" (has alway been that way with all my 4 LCD's )

Best 73
Jos (PA3CCE)

df8oe, 2016-04-13:
Hello Jos,

works for hundreds of users - so let's investigate...

Prepare something like "thin paper" where you can fix position of WF/SCOPE bar and position of Tuning Step Size.

Now boot into "Input Elements Test" mode. This is done by pressing & holding a button (except POWER or BAND-) during splashscreen.

Now test positions that were shown when touching on edges of WF/SCOPE-Bar and Tuning Step Size.

The positions must (!!) be within rectangular:
30/57 27/31  WF/SCOPE

and
26/35 and 39/43

Numbers are positions:
x(left)/x(right)     y(down)/y(up)

Touching is recognized within a box given by this coordinates.

I am nearly sure yours differ very much. At the moment there is NOTHING TO DO because changing to your positions make touchscreen "unworkable" for most other users.

Do not bother about "Touchscreen not available". If it givescoordinates it is working. I cannot reproduce this here.

vy 73
Andreas

PA3CCE, 2016-04-13:
OK Andreas, that is more or less what I did.

I did it again :
==========

1)  I marked the positions of the stepsize and the bar on the screen.
     (easy, because the protection  foil was still present).
2) Activated  "Input Elements Test".
3) Tapping on Stepsize marking with a stylus :
    Result : 
    x=29 , y=46 most (left position of marking).
    x=30 , y=46 (mid position of marking).
    x=36 , y=46 (most right position of marking).

Did the same thing wit the 2nd LCD, results IDENTICAL

4) Tapping on the Scope/Wfall  Bar marking with a stylus:
    x=06 , y=28  (Most left position of marking).
    x=30 , y=28 (Mid position of marking).
    x=57, y=28 (Most right position of marking).

2nd LCD IDENTICAL.

As said tapping the bar is working.
Tapping the stepsize has a problem :  target y=43,  measured = 46.

I think the Y-margins of the stepsize need adaption, due to parameter spread.

I hope this helps

Vy 73 de
Jos

df8oe, 2016-04-13:
I have 6 LCDs that are all identical, too: but not identical to your results.... All show "43".

As a "dirty workaround" I have raised borders. Actually daily snapshot is build with this.

As "soultion" I will implement touchscreen calibration next time :)

vy 73
Andreas

closed 2016-04-13 by df8oe

PA3CCE, 2016-04-13:
OK, now it's working but, same as the other features, not consistent;
mostly I have to tap several times to have it switching.
Calibration might improve things hopefully.

Thanks anyhow.

73 de
Jos (PA3CCE)

df8oe, 2016-04-13:
We will see :) I have no other candidate for testing - ists workíng fine for all others as it is now....

vy73
Andreas

df8oe, 2016-04-16:
Hello Jos,

Ihave started implementing touchscreen calibration. But I do not know anyone except you who needs this and can test this :( I would be very happy when you can help me by testing some "individual builds of firmware" that I send you via email.

With all of my LCDs calibration function does not change anything and I do not know anyone else beside you who can help ;)

If you want to help please contact me via email om@df8oe.de

vy 73
Andreas

df8oe, 2016-04-18:
Test result was that your touchscreen is running out of specs. I do ot know any other LCD that gets so bad results so I think it is defective. If it is a "new series" I must have some (!!) of them here to investigate new algorhythm.

Does anyone else have so much problems in running touchscreen functions?

Sorry - nothing to do at this moment.

vy73
Andreas