df8oe/UHSDR #250: sync am does not save correct tuning and cat does only work in 10 hz steps

DG9BFC, 2016-04-13:
if I hear a station in sync am and all is set correct (means last digit is zero and no beat note can be heard ... calibration made to have no beat note) ... and switch off ... and back on ... then the station has a beat note of around 10 hz or so (means it is off frequency!) ... if i just turn 1 hz up or down and then back to the zero that beat note is away ...

seems as the mcu does not store frequency correct or better say when switched on the calibration or last digit is not read correct ... if i just turn 1 hz it jumps that last 10 hz to right spot (to 0.001 or to 9.999) and the 10 hz hum is away

i have an idea where to search .... (or i found another fault!)

also cat commands do only turn in 10 hz steps ... means if i finetune in hdsdr the last digit (1 hz steps) then only 10 hz steps are sent via cat towards mchf ... and also backwards ... the mchf sends only 10 hz steps towards hdsdr via cat .... i thought that 1 hz is finest step ... but thats not the case

maybe that is also the reason why sync am is 10 hz off when switched off and back on?!?

that 10 hz fault is allways there in sync am after power off/on ....regardless if i select cat to run in sandbox or not ... (it is also off frequency if no cat cable is connected at all)

greetz

sigi

dg9bfc

DD4WH, 2016-04-13:
Hi Sigi,

good observation! I can confirm that in normal RX mode (I have not tested CAT). But it is not restricted to SAM, it is also the case in SSB!

My impression is, it has nothing to do with SAM, but with the init of the Si570/frequency which is a little off on switching on, but is absolutely correct the first time you move encoder 4.

It has nothing to do with warming up etc., because the frequency offset immediately goes away when you turn the encoder.

73 de Frank DD4WH

DG9BFC, 2016-04-13:
it is maybe also in other modes but i could not hear that 10 hz in ssb 
(maybe cause by filter selection) ... but i can clearly hear it in sync 
am ... have to test it also in other modes

ok ... anyway ... programmers will know where to search :-)

dg9bfc

sigi

ps ... cat does not know sync am ... but hdsdr knows ecssb with 
selecting usb/lsb or both

DG9BFC, 2016-04-14:
a hint for all users and programmers ...

when you connect hdsdr and open "rf front end and calibration" you see a "phase meter" (looks like a clock or compass rose) ... with using several different radio stations and finetuning the calibration in the mchf i can now be right on spot on nearly all stations (if not then the radio station is off!!!)

most stations are really right on spot ... but some are not (20 hz low or high) ... and you only can hear that in sync am (and see it on the phase meter in hdsdr)

sadly the cat does not know 1 hz steps ... so YOU have to finetune that last digit

greetz

sigi

ps ... is there another different cat mode that is well documented but knows finer steps???

db4ple, 2016-04-15:
Re CAT: probably there is one with 1hz. Someone needs to check the hamlib source code.
The other issue could be related to the frequency temp correction. At startup this is disabled, only after first temperature measurement a correction factor is used. This is only applied after you turn the dial.

DG9BFC, 2016-04-19:
after power off on the frequency is not correct (we already talekd about the dds not set correct after power on and before turning the dial)

now my idea ... after about 3 seconds after power on the voltage of psu is sent to display ... can we also add a dds check at that point??? then after those 3 seconds the frequency is right on spot without turning the dial ... frequency IS STORED correct at power off ... but it is not set correct after power on ... with a dds check point in the power up sequence that could be solved ...

db4ple, 2016-04-19:
Hi Sigi,

yes, the fix for the frequency at startup problem is along these lines. I was busy trying to understand that part of the mcHF (how timing related things are handled), so that we can do this is in a good way. However, this issue is more difficult than that. I also have to see how temperature vs. frequency drift is handled during tx (here due to PA heat changes may also happen, especially in digital modes with high power). I haven't yet checked if  this is corrected on the fly and if it would work properly.

df8oe, 2016-04-19:
I have implemented this "retune after delay" now. Please check out.

vy 73
Andreas

DG9BFC, 2016-04-19:
in daily of 9. 32 today??

or where to find??

df8oe, 2016-04-19:
After discussion in our team this is reverted. It is not a good solution at all - sorry.

vy 73
Andreas

DG9BFC, 2016-04-19:
no problem ...

i am sure sooner or later that will be solved anyway

greetz and 73

dg9bfc

sigi

DG9BFC, 2016-04-19:
can be closed ... is solved now

closed 2016-04-19 by DG9BFC