df8oe/UHSDR #259: snap function

DG9BFC, 2016-04-14:


DG9BFC, 2016-04-14:
after pressing snap several times ... there is a green line added between 1hz and 10 hz digit (across these 2 numbers and half way down into the spectrum screen)

DD4WH, 2016-04-15:
Hi!

strange! The snap function does not deal at all with any graphical
representation . . .

I use the snap function 100 times every day and have never encountered
this. Can you give us more detail on your settings? Please also use the
latest firmware (you probably already did so).

73 de Frank DD4WH

df8oe, 2016-04-15:
I cannot reproduce here. Please give exact description of settings and what to do to get this. (Spectrum / WF used, what mode, what bandwidth etc. and test if setting affects issue).

vy 73
Andreas

DG9BFC, 2016-04-15:
hello Frank

yes it is strange ....

i had it 4 or 5 times here

needed around a dozen presses of snap or so ...(maybe more??)

but i had it only a few times ... and it takes a few dozen tries to 
reproduce it ...

i made a picture with my digicam if that helps

dg9bfc

sigi

DG9BFC, 2016-04-15:
it is also not easy to reproduce here (takes a few dozen tries)

i made a picture with my digicam (better say 2 of them showing both faults)

sometimes that fault does happen left side at the spectrum line (upwards 
from normal green line between 10 and 1 mhz digit)

and sometimes on right side (between 1 hz and 10 hz digit)

.....

and it does happen VERY SELDOM (means you need many tries to reproduce) 
... sometimes 20 or 30 tries needed

but there is that "software glitch" ... (can not explain it better)

greetz

sigi

dg9bfc

ps did you get the pictures??

DD4WH, 2016-04-15:
Thanks a lot for the photographs, that helps!

I observed a similar problem (probably same bug!) in the middle of the
spectrum display. However it is mainly hidden behind all the
black/grey/white pixels.

Tune in a station and observe the spectrum display at the tune frequency.
Sometimes there are pixels that are not plotted in white, although they
should be there. But they stay black. The pattern is the same as your green
bug graphics.

Probably a bug in the spectrum display driver, that is somehow mislead by
the intense and longlasting calculations for the snap FFT. In that same
routine the samples for the spectrum display FFT are being collected, seems
to be a timing issue!? But why do the lines occur at arbitrary places?

This will not be easily fixed, because that driver is particularly complex
with all the AGC loops and lowpass filters built in! Clint has done a
fantastic job in programming that, but for me it is not at all easy to
interpret.

We will keep this in mind, it will be on the fix list.

73 de Frank DD4WH

DG9BFC, 2016-04-15:
i do not know if the photos do help ...

but i know that this fault is seldom ... not easy to reproduce  ...

so i though i make these pictures to "prove" it is there

i only have it if i press the snap a few dozen times or so ... and only 
when you type fast ... so a timing issue is a possible reason ...

maybe it is when i press on snap while it is still doing calculation 
from pressing before??

i dont know .....

... by the way .. why does the snap function does not find the sweet 
spot (0hz difference between calculated carrier and real signal) ...??? 
i ALLWAYS have to finetune the last digit (sometimes 10-20hz off ... 
sometimes only 2-4 hz off)

greetz

sigi

dg9bgfc

df8oe, 2016-04-15:
On a perfect, undisturbed and clean reception signal it *would find* exact carrier :)

Each fading, ticking etc. does calculation unprecise.

vy73
Andreas

DD4WH, 2016-04-19:
### you wrote:
... by the way .. why does the snap function does not find the sweet spot (0hz difference between calculated carrier and real signal) ...??? i ALLWAYS have to finetune the last digit (sometimes 10-20hz off ... sometimes only 2-4 hz off)
###
It is an interpolation method. If you would use only the FFT bins, you would have an accuracy of +-47Hz (the bin-width), more is theoretically not possible. Now you take a quadratic interpolation with three points of a quadratic parabel standing on its head. And that trick gives you +-2Hz. I think that´s great for this simple method! Compare that with HDSDR which uses a toolchain for the carrier search alone, that has twice the complexity of all the mcHF audio and display chains together (with DSP notch/DSP NR switched ON/largest bandwidth in AM/large waterall display)  . . . . ;-)

We are still searching for the graphic bug, don´t worry!

73 de Frank DD4WH

closed 2016-04-19 by df8oe

DG9BFC, 2016-04-19:
snap funtion now much better

after a few tests i can say it is ok like it is now (yes i know we have 
not that much cpu power)

i do not worry about that graphic bug ... i just reported that it is 
there (SLEEPING SOMEWHERE IN CODE .. HI HI) ...

maybe it is just a timing thingy and it will be solved sooner or later 
on its own (when all functions are cleaned and running well) ... i know 
that we had some trouble with nr and also that was solved with cleaning 
the code ... so my guessing is that i do not have to worry ... just have 
to wait a few days

mchf is getting better every version (sometimes in small and sometimes 
in larger steps)

and i am sure also the graphical thingy will be solved (there are other 
graphic faults now coming from light spectrum and from big spectrum 
adding but also those will be solved soon for sure)

....

issue closed