df8oe/UHSDR #267: REQUEST: Voltage depenended power adjustment

dm6tt, 2016-04-17:
The output power depends directly from the supply voltage.
Since the supply voltage is measured, the output power could be adjusted accordingly.
This would lead in an independed choosen output power from the power supply.

This would be a nice feature form OMs that run the mcHF on different supplies (shack, battery pack, car).

73, Marcus, DM6TT

df8oe, 2016-04-17:
Hello Marcus,

I have had this idea in July last year, too. But I investigated that:
- function is not linear
- differs to modifications
- differs in bands

So it will be *a very huge and complex* work with again dozens of EEPROM configuration places.

Because of this I decided *not* to implement it.

vy 73
Andreas

db4ple, 2016-04-17:
If it cannot be put into a (set of)  formula(s), I would not want to implement it. 

If it can be put into formula(s), someone could provide these here. And we can integrate that.
That would be nice to have. 

Formula input would be the adjustment settings per Band, the available voltage. Possibly the voltage drop (i.e. internal resistance of the power source) should be considered to. Especially for not so well charged batteries connected via not so good cables, voltage drop can be significant in higher power TX. However, getting that value is a little more tricky.

73
 Danilo

df8oe, 2016-05-13:
too much work...

vy 73
Andreas

closed 2016-05-13 by df8oe