df8oe/UHSDR #291: spectrum line behind signal in non light but big mode and no grid

DG9BFC, 2016-04-22:
if i switch to big mode (means the grey bar off for bigger viewing area) ... then the spectrum line is hidden behind the signal

and in big mode the spectrum grid is off ...

... when i switch to normal mode and back to big ... then the centre line is in front of the spectrum and the grid can be partly seen (it is where a signal "was") ... after next power off/on the grid is away and the centre line again behind spectrum

and light view with normal height (means grey bar on) the signal still draws inside that grey bar

so ... normal mode works like it should .... light view works only in big mode (grey off) or it draws in the grey bar

and normal mode in big view ... has no centre line and only a partly grid

df8oe, 2016-04-23:
some parts fixed:
- grid now is completely drawn in "spectrum standard" --> BIG
- centre line of spectrums always is in front

2do:
- "spectrum light" always uses vertical range of BIG (and draws in bar when amplitude is high)
- "spectrum standard" only uses vertical rande of NORMAL(and never uses upper millimeters)

vy73
Andreas

DG9BFC, 2016-04-23:
grid is still not completely drawn (the lower end is not shown correct)

if no signal is present and you switch on then the grid is there (as is the centre line) ... but a signal (or some signals) are "deleting" the grid

before in older version the signal was drawing that grid ... now it is reversed (signal wipes grid out)

but grid is not permanently there

so .. it is partly fixed ( the centre line) but grid is still not right (but i am sure also that will be fixed)

73 sigi dg9bfc

DG9BFC, 2016-04-23:
vertical lines of grid are permanent ... but horizontal lines are whiped 
out from a signal

vy 73

sigi

df8oe, 2016-04-23:
horizontal redraw fixed

vy 73
Andreas

DG9BFC, 2016-04-23:
and issue closed :-)

73

sigi

closed 2016-04-23 by DG9BFC