df8oe/UHSDR #292: VFO switching

dl2ki, 2016-04-22:
Hi,

if I set VFOA, LSB, and then toggle to VFOB and there turn on CW, CW will remain at the switch back to VFOA. LSB is however displayed in the mode display.

It makes sense if the mode would be preserved when switching the VFO's. You can then use the two VFO's separately for the SSB- and the CW subband.

73, Wolfgang
DL2KI

DG9BFC, 2016-04-23:
can also be seen in other modes

example .. it would be nice to use am on vfo a and sync am on vfo b ... and switching between sync am and normal am could be done with the vfo a/b button
 (without the need to switch trough other modes)

73 sigi

dg9bfc

df8oe, 2016-04-23:
fixed.

vy 73
Andreas

closed 2016-04-23 by df8oe

DG9BFC, 2016-04-23:
not fixed right (filter is not changed)

vfo a/b changes mode ... but in the filter box the filter from old 
setting is shown

example you have sync am in vfo a with 4.2 kc

and and in vfo b you have am with 4.6 kc

toggling vfo a/b and the modes change (seen in mode box above mhz digit) 
... but not the filterbox

73

sigi