df8oe/UHSDR #300: VFO switching

dl2ki, 2016-04-23:
Hi, Andreas,

something is still wrong.

If CW is involved, the frequency will be shown in blue.

Example: VFOA = LSB, CW = VFOB
When switching from VFOB on VFOA SSB is incomprehensible. LSB must be set again.

73, Wolfgang
DL2KI

df8oe, 2016-04-23:
I cannot reproduce here. Frequency always is in WHITE, modes and filter bar are switched correctly.

???

vy 73
Andreas

dl2ki, 2016-04-23:
Video is on the way via email.

73, Wolfgang
DL2KI

df8oe, 2016-04-23:
I cannot reproduce at all.

y 73
Andreas

db4ple, 2016-04-23:
Same here.
7076 khz VFOA LSB, VFOB CW, Switching back and forth works.

DG9BFC, 2016-04-23:
so it does here ... switching works fine

73 sigi dg9bfc

dl2ki, 2016-04-24:
Hi,
with the firmware "24.04.2016 18:04:42" works the VFO switching now appears correctly. What was the problem?

73, Wolfgang
DL2KI

db4ple, 2016-04-24:
If I would know that exactly... I cleaned a number of redundant calls to initializes this and that. Most of these have been there for quite a while, but it also contained 2 request for frequency almost back to back. Maybe this caused trouble, but on the other hand,  you got a "steady" blue, an situation for which I still have no idea how to get into that.
 
Anyway, I simplified the code to a point where I think, I know what is going on, and it looks ok to me at least now. Since it also fixed your issue, I am happy but also somewhat clueless.

Let me know if this happens again, and I'll dig deeper into it with your help.

Many thanks for reporting. Andreas, can you please close that issue, I think Wolfgang will agree to this.

73
 Danilo

DG9BFC, 2016-04-24:
why after power on the filter is in orange?!?

73

sigi

df8oe, 2016-04-24:
I never have had noticed the mentioned issue - on none of my mcHFs.

But if it is gone: everything ok now :)

vy 73
Andreas

df8oe, 2016-04-24:
Filter in orange is a new feature. Now selecting filters via encoder is default. Or do you like old method more??

vy 73
Andreas

db4ple, 2016-04-24:
Honestly, I am more for the old way. It is easy enough to select a filter anyway but I more often switch the input line than the filter. YMMV.

@Andreas: Technically switching back  is just changing the default of filter_path_change back to false.

73
 Danilo

DG9BFC, 2016-04-24:
ah ..ok i see ...

ok on that ... i had not noticed the encoder function behind it ... and 
if it is on by default ... so be it

(i though it was a bug ... grin)

but after power up i thought that the default would be what you had with 
power off ... and if you have "3 filters selected for am" ... and power 
off with filter 4.8 am in blue ... that power on also same would be set

yes ... same filter is set ... but not if encoder is engaged

would it be possible that always last setting is stored ??

so if encoder engaged it is again on after power on ... but if user 
prefers the simple setting (only 3 filters usable with button) and 
encoder is off ... it should be also off after power on

(you safe last settings in eeprom in power off ... right??? ... so just 
also store the encoder on/off in eeprom)

73...... sigi

DG9BFC, 2016-04-24:
i change the usb iq or analogue audio several times ... depending if i 
use mchf with any digital modes soft ... or with hdsdr ...

for filter is have set a few that i use in filter settings ... and 
others also can be reached

dg9bfc

sigi

df8oe, 2016-04-24:
Old way is preferred, Danilo...

vy 73
Andreas

closed 2016-04-24 by df8oe

DG9BFC, 2016-04-24:
both ways should be possible

only thing is ... mchf should REMEMBER setting of orange / blue with 
power off

and should use same state when power on

.......... then all is solved and user can select what he prefers ....

.......

  andreas ... can you add a button below the filter box (inside the 
box)?? (like in signal meter or in the digital box ... grin)

would be another nice touch function

tapping selects new filter bw (either only 3 or 4 ... or all selectable  
depending if box is orange or blue)

.... and in parallel to the (filter) button that we have now (for the 
"GRMBL" non touch users) ...

73 ... sigi