df8oe/UHSDR #305: Menu - Hardware info does not show active touchscreen

DL3NCW, 2016-04-24:
Have made touchscreen mod and touchscreen is working like switching meter, choose signal in spectrum scope etc.
If going into menu - hardware info, it shows "touchscreen n/a" and touchscreen does not work in menu.

df8oe, 2016-04-24:
I have heard this from some OMs but I cannot reproduce it here and I have no idea what can go wrong. Sorry - no solution at the moment :(

In Menu mode touchscreen gets completely different functions - at the moment for debugging purposes (not user documented).
vy 73
Andreas

dl2ki, 2016-04-24:
Hi,

the same behavior here.

73, Wolfgang

DG9BFC, 2016-04-24:
same here

touch function works but is not shown in hw info

72 sigi

df8oe, 2016-04-24:
I see. It worked since months correctly and so I do not have tested it again.

Since ??? it does not work. I will take a look.

vy 73
Andreas

DG9BFC, 2016-04-24:
same here ... i have not seen in hw info for many weeks (cause tp 
function worked i though in hw info it is also shown correct) ...

... sure you will find it :-)

73

sigi

df8oe, 2016-04-24:
so it is ;)

fixed.

vy 73
Andreas

closed 2016-04-24 by df8oe

DL3NCW, 2016-04-24:
Yeahh Andreas you are my hero !