df8oe/UHSDR #306: Frequency translation does not work anymore

DL3NCW, 2016-04-24:
despite switching frequency translation, the indicator line always stays in the middle of the
spectrum display.

DD4WH, 2016-04-24:
Wrong title ;-) you mean the INDICATOR LINE, not the frequency translation per se !

Please give more info. I cannot reproduce this here.
 73 de Frank

DL3NCW, 2016-04-24:
Ok, when frequency translation is of he indicator line was in the middle. When switching frequency indicator on, depending whether +/- 6/12 khz the indcator line moved to the left or right side of the spectrum. Now with the lates daily the indicator line stays in the middle.
I dont know if it is a bug or feature ? :-)

73 de Joachim

DD4WH, 2016-04-24:
Do you know the MAGNIFY-mode? This gives you a spectrum display/waterfall of double size. If you have switched on this mode (which you probably have), the indicator (and your frequency) is ALWAYS in the middle. Find Magnify in the display menu and switch it off/on, then you will see what happens.

DD4WH, 2016-04-24:
if that works, please close the issue.

DL3NCW, 2016-04-24:
You are right, this was on. I have not switched it on intentially...

closed 2016-04-24 by DL3NCW

df8oe, 2016-04-24:
I think you have activated it via touchscreen.

If you touch left half of the "Waterfall/Spectrum Scope" bar you toggle magnified on/off and if you touch right half of the bar you can toggle waterfall / spectrum. Knowing your unprecise touchscreen youhave activated it by touching on waterfall / spectrum (your tc tells xoordinates some millimeters downside).

vy 73
Andreas