df8oe/UHSDR #307: Switching from Digital - Freedv1 - etc

DL3NCW, 2016-04-24:
By using touchscreen one can switch from Digital to Freedv1 and so forth.
The same happens if one pushes "DSP-OFF"

DD4WH, 2016-04-24:
as far as I know, that is absolutely intended!? That is the preparation for the digital modes.

DL3NCW, 2016-04-24:
You are right, it is not yet implemented.
But one does not want to switch by also pressing on DSP-OFF as this button should do something else

df8oe, 2016-04-24:
Your touchscreen is *extremely out of specs*...

Touching in DSP should not affect digital mode settings. I know one other OM from Yahoo group whose LCD is absolutely out of specs - other direction. I am not sure that this can be solved with a touchscreen calibration caode ( I am just working on it). Ithin youhave to replace LCD...

vy 73
Andreas

closed 2016-04-24 by df8oe