df8oe/UHSDR #314: Colour of the frequency display

oz8gw, 2016-04-25:
I wonder why the colour of the freq. in the display is blinking blue-coloured when I turn the freq. wheel fast in the later "binaries of actual development as 'daily snapshots:' from the "McHF-GitHub" site.
Vy 73 de OZ8GW, Leif.

df8oe, 2016-04-25:
Hello Leif,

thats a new feature :) If you turn dialknob very fast so that two tune commands were send to SI570 in less than 20ms SI570 will crash. So only possibility is to suppress these fast commands and onlysend those which follow > 20ms. In former firmware this was not indicated so that there was no possibility to see if SI570 tunes according to turn vfoknob. Now digits change to blue if you turn too fast and indicates SI570 cannot follow.

vy73
Andreas

oz8gw, 2016-04-25:
Thanks for the answer.
Is there a place where I can read about the new updates?
Vy 73 de OZ8GW, Leif.

df8oe, 2016-04-25:
At this moment there is actually NO place where all new features are listed. But I will start the next days new page in our WIKI where I describe all additional functions which were implemented since Clints last release.

vy 73
Andreas

oz8gw, 2016-04-25:
Sounds good, Leif.

closed 2016-04-29 by df8oe