df8oe/UHSDR #315: Band and VFO switching.

dl2ki, 2016-04-25:
Hi,

in both cases, it might be useful to save a fixed set of parameters, which then would be available after band- or VFO-switching again. e.g. Frequency, mode, VFO A / B, tuning step rate, DSP parameters, ...

The parameters could be determined perhaps by a small survey.

73, Wolfgang
DL2KI

df8oe, 2016-04-25:
There *is* already a set of fixed parameters saved! Frequency and mode are saved when you change bands or vfo and are restored when returning. Actually tuning steps and DSP parameters are not saved and I am not sure this is possible or possible partwise (band-specific?!) without destroying complete configuration - what we would NOT do.

vy 73
Andreas

dl2ki, 2016-04-25:
Is not so important, was just an idea.

73, Wolfgang

df8oe, 2016-04-25:
ok ;) Let's see. There is ONE free byte in band storing and we will decide carefully what to do with it...

vy 73
Andreas

closed 2016-04-25 by df8oe