df8oe/UHSDR #316: Band when switching between VFO A and VFO B

DL3NCW, 2016-04-25:
With the latest daily of today i found the following:

VFO A tuned on 20 m for example and VFO B tuned on 40 m
When switching from VFO A to B everything looks ok but when 
switching back from VFO B to A, VFO A is tuned on 20 m.

Is it a bug or a feature ?
I expect to be on 20 m again if switching back from VFO B to A.

73 de Joachim, DL3NCW

df8oe, 2016-04-25:
Band splitting was never intended or is it actually supported. What happens when tuning vfo a and vfo b to different bands is "by chance" and can differ between firmware versions. It is recommended to tune both vfos to the same band.

vy 73
Andreas

DL3NCW, 2016-04-25:
Thanks Andreas,

i remember that it once worked.
I dont need "Band splitting"(for tx) it would make it just easier to
jump between frequencies, i.e. broadcast and ham bands.

73 de Joachim

db4ple, 2016-04-25:
I can confirm that band splitting was not working properly for the TX case until now (pull request #318) as the power factors were not set properly according to the real TX frequency which can kill at least your PIN diode if transmitting with too much power into a high SWR antenna by accident.

I did some work to clean up this code and make it work in the last days. There is also now some protection from doing TX on "out of range frequencies" in split mode (and of course in normal mode as well).

db4ple, 2016-04-25:
Joachim, if you confirm that everything is working with the next build including my pull request #318 Andreas can close the issue.

DL3NCW, 2016-04-26:
Sorry doesnt work for me. Still the same issue described above

df8oe, 2016-04-26:
Sorry works for me...
I cannot reproduce.

vy 73
Andreas

DL3NCW, 2016-04-26:
Just to clarify: I would like to have reception of i.e. 20 m on VFO A and 40 m on VFO B.
Switching now from A to B makes VFO B to receive also on 20m not 40 m
Could it be that last frequency is not stored on the VFO's or last frequency is not restored ?

db4ple, 2016-04-26:
Hello Joachim,

now I got it: You are on 40ms and use the VFO B. You switch to VFO A and then switch to 20m. You expect VFO B to be still on 40m since VFO B was on 40m when you left it.
The interaction between VFO  and Bands does not work like this in mcHF (never has).
Bandswitch switches boths VFOs to the same band at the same time. The mcHF SW currently a single active band. 
What does work  now, is to tune to a frequency outside this band in split mode with VFO B and you're able to transmit on that out of band frequency. It is even stored but in the VFO B 20m memory.

This is strange but this it how it was and still is. So no bug, but may an opportunity for improvement.

df8oe, 2016-04-26:
Here is what I tried:

1) switch to 20m and adjust VFO A to a frequency on 20m, e.g. 14.224, mode USB
2) now switch to VFO B and tune frequency to 40m, e.g. 7.168 and switch mode to LSB

When toggling VFO A and VFO B by pressing VFO button VFOs change accordingly to tuned frequencies and modes.

What is the problem? How can I reproduce an issue?

vy 73
Andreas

DL3NCW, 2016-04-26:
Andreas, that is exactly what is not working for me.
When i do like that and switch back to VFO A i am on last tuned frequency on 40 m of VFO A...
Shall i reset the serial eeprom ?

db4ple, 2016-04-26:
Hi both of you:
How do you tune to the other band: 
a) Turn the dial know until you reach the band or 
b) by using the band -/+ and then tune in the new band to the  frequency.

df8oe, 2016-04-26:
No. That cannot deal with serial EEPROM.

I think it is an operational error. You haven't set MemFreq Lim Disable to ON in config menu ;)

So no frequencies outside band can be stored (nor temporary or in EEPROM)
vy 73
Andreas

DL3NCW, 2016-04-26:
MemFreq Lim Disable is OFF.
I change the band by +/-

df8oe, 2016-04-26:
If you want to tune both vfos to different bands you NEVER SHOULD TOUCH BAND KNOBS. Only dial knob. By changing bands you *always* change BOTH vfos to correspondent band storage place.

vy 73
Andreas

DL3NCW, 2016-04-27:
I dont consider this as a practical solution.
Think of changing band from 80 to 10 m by VFO Knob...
It only takes some clicks on the +- button.
Why you dont implement two separate working vfo's like on
all the modern transceivers ?
Is there any reason ?

vy 73 de Joachim

df8oe, 2016-04-27:
Two reasons:

1) I don't see that others need this change, too. Noone has asked for such a feature and I would not like it. It would destroy easy split mode stored for each band I intensively use.

2) The existing storage structure does not allow this. There must be huge changings with the lost of complete configuration data when it would be implemented.

no bug - closed.

vy 73
Andreas

closed 2016-04-27 by df8oe