df8oe/UHSDR #330: Tune Power

dl2ki, 2016-04-28:
Firmware Stand vom : 28.04.2016 11:39:28

Hi,

from my point of view there is no tune-power in the 60m, 15m, 12m and 10m band.

73, Wolfgang
DL2KI

df8oe, 2016-04-29:
Hopefully fixed (only reproducable on some mcHFs - not on one of our testing devices)

vy 73
Andreas

dl2ki, 2016-04-29:
Hi,

trying out the current DS, I have found that the missing tune power for me depends on the frequency.

- 3511kHz no power, even as 7.011kHz, 10 111kHz, 14.011kHz
- from 3512kHz up, output power OK, even as 7.012kHz, 10.112kHz, 14.012kHz

73, Wolfgang
DL2KI

db4ple, 2016-04-30:
Hi,
thanks for reporting this. Issue was stricter checking of TX power values. Unfortunately not the TX frequency was used but the IQ center frequency (which can offset +/- 12khz). And depending on your settings 3511 - 12 -> 3499 -> outside ham band -> no TX power settings -> no power. You get the idea. Fix is being prepared.

To test:
If you change the translation offset, the frequency will change to 3505/3506 or to the top of the band.

dl2ki, 2016-04-30:
Hi Danilo,

fine. My "RX/TX Freq Xlate" setting is "+12kHz".

73, Wolfgang

db4ple, 2016-04-30:
Fix as been test (works for me) and merged.

dl2ki, 2016-05-01:
The fix works here also.
Therefore, the issue can be closed.
73, Wolfgang

closed 2016-05-01 by df8oe