df8oe/UHSDR #332: Request: CW sidetone without transmit.

Rescue9, 2016-04-28:
Currently, the only way to generate a sidetone is to actually transmit. While this can be done at the .5w fairly easily when hooked up to an antenna / dummy load, there needs to be an option to generate sidetone without transmitting anything. this facilitates "on the go" cw practice and learning.

DG9BFC, 2016-04-29:
hmmm ... that could be done with setting tx off in menue but sidetones still generated when pressing paddles in cw mode (just as an idea how the user can set that up in his mchf)

73 sigi

Rescue9, 2016-05-12:
I may have found another quirk. CW sidetone doesn't seem to be generated through USB audio out, nor through the audio out jack. This would also be helpful.

df8oe, 2016-05-13:
I cannot see why sidetone shall come out of line and USB audio, too. I never missed this and cannot get an idea how to use this. We will put on our 2do list generating tone when tx is disabled.

vy73
Andreas

df8oe, 2016-05-31:
This is realized since 05/31/2016.

closed.

vy 73
Andreas

closed 2016-05-31 by df8oe

df8oe, 2016-05-31:
No - not only LED is switched off.

In function "SwitchTxRx" GPIO which enables mixer and pA bias is disabled, too so that really no transmissionis done.

vy 73
Andreas

DG9BFC, 2016-05-31:
hmmm i have an idea what the other user (rescue9) wants with sidetone generated on line out or usb audio ... he wants to send his "cw practise" to a digital decoding soft (or record it with a wave player)

Rescue9, 2016-05-31:
Yes. Having sidetone generated over USB or line out would greatly help out with CW PRA rice. Especially when this happens online in a group such as CWOps puts on.

db4ple, 2016-06-01:
I have figured out how the sidetone you listen to on the mcHF is generated in the first place now. I think it should be fairly easy to get a USB sidetone as well and also lineout (this is even easier).
What I don't know is, how popular such a feature would be. I.e. shall we do this generally or do you need to enable that in the menu? I would tend to the latter option. Rescue9, could you be so kind to create a new issue for that task and start a discussion?

DG9BFC, 2016-06-01:
what about this ... you block tx with pressing tune long ... now when 
pressing paddles the sidetone is still generated ... but no power comes 
out of antenna port

dg9bfc

sigi

Rescue9, 2016-06-01:
Instead of long-pressing tune, how about long-pressing the output power button?

 

Rescue9

df8oe, 2016-06-01:
From my side this is finished.

By disabling TX (in menu or by long pressing tune) in CW now sidetone is generated - but no rf leaves mcHF. Further additions like sidetone via line out or USB will only be possible with much more changings and I do not know if this is neccessary...

vy 73
Andreas

Rescue9, 2016-06-01:
Then I'll create a new issue and copy over some of the pertinent information. Thanks!