df8oe/UHSDR #336: Reworked algorithms and code structure for Si570

dl2ki, 2016-05-01:
Hi,

i hear a QSO round on 1.211.500Hz in USB. An assignment to a regular frequency I can not do at the moment.

73, Wolfgang
DL2KI

db4ple, 2016-05-01:
Can you hear it on the normal frequency as well or is it only on the "new" frequency?

DD4WH, 2016-05-01:
Hi Wolfgang,

good observation! But that has nothing to do with the Si570 algorithm.

It is a normal feature of I/Q SDRs:

The Tayloe detector/quadrature sampling detector receives not only at the desired frequency, but also at odd divides of the receive frequencies. That means you can receive frequencies that your LO cannot reach, thats called undersampling, I think.

Not sure whether this is correct, but try the following formula:

F(real) = n * F(mcHF) - sampling rate
n is 3 or 5 or 7, etc.
in your example:
F(real) = 3 * 1211.5kHz - 48kHz = 3586.5kHz 
Right in the 80m- Band!
You will not receive multiples of 5, 7, 9 with the mcHF because of the bandpass and lowpass filters in the RX path, BUT at 1211.5kHz, the 80m Lowpass and bandpass filters are active, thats why you receive that QSO! Try to verify that by switching to 80m. It could be that you have to add/substract the IF frequency, if you have frequency translate enabled, but I am too lazy at the moment to incorporate that into the formula!

So, your example shows that we definitely need other bandpass/lowpass filters, if we want to extend the mcHF to other bands such as 160m etc.

So this behaviour is absolutely normal and present in every mcHF!

Have fun!

73 de Frank

 

-->

DD4WH, 2016-05-01:
Oh, I missed that you gave the REAL frequency of the QSO!

So the right formula seems to be (PLUS instead of MINUS !):

F(real) = n * F(mcHF) + sampling rate

in your example:
F(real) = 3 * 1211.5kHz + 48kHz = 3682.5kHz

db4ple, 2016-05-01:
Hm. Down in that frequency area, strange things are happening. If I go below 1455khz, the tune frequency observed by my DSO goes up instead of down and the output looks anything but clean (but this is most likely due to the "missing" LPF/BPF for this frequency range).
Closer to 160m it looks much better, though.

db4ple, 2016-05-01:
If I transmit on 3682.5 I receive the signal also at 1243.5 * 3 - 48 = 3682.5

DD4WH, 2016-05-01:
Hi Danilo,
yes, if you use an IQ SDR without the right bandpass/lowpass filter, you have all kinds of mirror frequencies, odd multiples/divides that enter your RX path. That´s why I have a very narrow manually tuned preselector in front of my Teensy SDR to eliminate all that.

db4ple, 2016-05-01:
Anyway, can we conclude that everything is working here with its abilities (in that respect)?
So Andreas can  close this one.

DD4WH, 2016-05-01:
hmmm, the transmit path also has a quadrature sampling exciter QSE, so that should have the same properties as the RX path with the quadrature sampling detector QSD.

I think we could close this issue, because to my best knowledge this is a general property of IQ SDRs.

A VERY good explanation can be found in part 1 and part 4 of this series of papers:

https://sites.google.com/site/thesdrinstitute/A-Software-Defined-Radio-for-the-Masses

closed 2016-05-01 by df8oe