df8oe/UHSDR #338: starting cat switches vfo to c

DG9BFC, 2016-05-01:


db4ple, 2016-05-05:
Have to check that. Blue VFO color should not stay, even in CAT mode.

db4ple, 2016-05-06:
Currently I cannot reproduce the blue color, is it only in HDSDR (haven't tested this yet) or in all CAT tools (I tested wsjtx and fldigi, no problems).

Thanks, 
Danilo

DG9BFC, 2016-05-06:
when i start hdsdr it switches to blue ... and also when starting 
commander (to use with multipsk) ... did not check other soft yet

dg9bfc

sigi

DL3NCW, 2016-05-07:
Also happens occasionally to me if i use HRD V5.x transfering
a frequency from the database into the mchf

db4ple, 2016-05-13:
We disabled the blue digits for now. However, we did not look into the root cause for the blue color staying. However, if it stayed it meant that the desired frequency was set. Otherwise color would have changed. So we may close this issue from my point of view or keep it.

df8oe, 2016-05-13:
never could reproduce...

vy 73
Andreas

closed 2016-05-13 by df8oe

DG9BFC, 2016-05-13:
with a version that has the blue digits active ... it was always with 
power on of hdsdr ...
every first cat switched to blue

turning dial or change qrg in hdsdr and numbers are white again

cause you now have the blue numbers not active i do not have that 
"fault" anymore ... it is maybe a small software glitch or similar thing 
... seems as polling qrg from hdsdr does switch to blue ... but as 
reported ... now i do not have it in this version cause you disengaged 
the blue numbers ...

greetz

sigi

ps ... i could show it to you with an older version ... just in case you 
wanna look for it