df8oe/UHSDR #339: REQUEST: bass and treble box permanent

DG9BFC, 2016-05-01:
now the button switches the box for bass and treble on and off ... what about making these boxes permanent (but only colour change what function is enganged) ... so the button press does only change between the boxes (rf gain dsp nb bass treble etc) ... but they stay on always
colour in text box change from grey to black and numbers from white to red when function selected with button press .... what do you think??

side note ... grey colour in treble and bass box numbers are a bit darker grey as the grey coulour in dsp or rfc gain box ... please change to the lighter grey
those numbers change to black (invisible) when seen from left side while the lighter grey changes to a darker grey and can still be seen

(another side note or a thank you to programmer)
if you lower the low tones (bass to -10) ... and lift the higher tones a bit (treble +10) ... then the noise reduction does not sound so bad (it sounded a bit to low and mumbled before) ... means now it is better usable ... thanks for adding the tone controls of bass and treble ... those differences can best be heard with a bigger active speaker

df8oe, 2016-05-02:
The boxes conflict with display of output power in mw...

vy 73
Andreas

DG9BFC, 2016-05-02:
i have an  idea for better display usage

first ... delete "vcc" boc (only show voltage in box below)

let all other boxes drop down one box size

now the blue boxes are a traight line with the grey bar (or with upper 
red line if  big mode selected)

and you have gained a lot of room for the grey set boxes ...

display power in power settings box .. in rx show setting in white  and 
in tx show measured power in red or orange ... (if mw power is engaged 
in menue show mw value ... if not show watts value) ...

with deleting "vcc" box ... and showing power in power box (wherelse 
would you search measured power if not in power box??)

you now should have enough room for bass and treble permanent ... and 
peak/manual notch below bass and treble with changing depending on 
usage/mode

...

imagine a nonmchfowner on fieldday ... he would know what that box means 
if shown in white while rx and in orange/red while in tx (power box) ... 
so no need to let an otherwise needed area free for display power (in 
mw) ... better make bass and treble permanent ... it is doable this way 
... and we do not waste screen area for a useless "vcc" box"

...

(yes i know it is a lot of work .... but the outlook is much nicer then)

dg9bfc

ps do not forget to change colour of the numbers in these boxes to a 
lighter grey like in the boxes above

DD4WH, 2016-05-02:
Good idea! But will take some time to discuss and implement.
Grey color has been changed already.

73 de Frank

DG9BFC, 2016-05-03:
i know that it will take some time to implement that idea ... but ... if 
we can safe some room for the bass and treble display (and notch/peak) 
with deleting a useless boc (vcc) ... yepp then i would prefer to have 
the bass and treble permanent and no vcc box anymore (voltage should 
still be shown in box below) ... when a nonowner (nice word) sits in 
front of an mchf ... and sees 12.5 V ... then he knows what that means 
... right?? ... so vcc is useless waste of room on the tiny display

the mchf now is nearly self explaining ... and so it will be after the 
gui modding ...

....

the upper end of the blue boxes in the same line as the grey bar (or 
with upper end of spectrum in big mode) ... that sure looks a lot nicer 
when no box "sticks out" on left side ... :-)

.....

thanks for changing colour

......

greetz

sigi

ps where is the "like it" button??????? ... thanks to all programmers 
for their work

dl2ki, 2016-05-07:
Hi,

question:

Are the settings needed permanent for bass, treble, peak/notch-filter? Therefore the question whether these display boxes in the small display space must be permanently visible?

The heading of VCC is certainly not required. You could provide the background of the value even with a color.

73, Wolfgang
DL2KI

DG9BFC, 2016-05-08:
i would like to see those boxes permanent

that was a request .... no need to follow my idea ... but i was guessing 
also others would like that
73 sigi dg9bfc

df8oe, 2016-05-10:
conflicting opinions - closed for now.

vy 73
Andreas

closed 2016-05-10 by df8oe